r-lidar/lidR #1: error in LoadLidar()

ptompalski, 2016-03-22:
When loading a pointcloud with LoadLidar() following error occurs:
```
Error in as.data.table.matrix(xi, keep.rownames = keep.rownames) : 
  4 arguments passed to .Internal(nchar) which requires 3
```

Jean-Romain, 2016-03-22:
Hi,

Please check and report your:
- R version (mine is 3.2.2)
- data.table package version (mine is 1.9.6)

I found 3 references with this error: here https://github.com/joey711/phyloseq/issues/564 and here http://stackoverflow.com/questions/31142280/unusual-error-in-setkeyv-with-data-table and here http://grokbase.com/t/r/r-help/156m0j2nfm/r-data-table-package-problem. It seems that the problem came from the R version.

If updating R solve the problem I will change the DESCRIPTION file.

ptompalski, 2016-03-23:
R version 3.2.2
data.table 1.9.6

After updating R to 3.2.4 problem does not occur.
This is strange and I am not sure what causes that.

Jean-Romain, 2016-03-23:
If the problem comes from R, it's out of my skills. 

Here with the same version than me (linux) and the same than another user (windows) you have the problem. You updated to a greater version than the one we are using. I can't modify the DESCRIPTION file to impose a greater version than mine because it works on 3.2.2. The problem comes from somewhere deeper than what I can investigate.

ptompalski, 2016-03-23:
I will try this on several other machines and report back. Hopefully this was a single issue on my laptop.

Jean-Romain, 2016-03-23:
Works also on R 3.2.2 on Windows platform on a friend's computer. We made 7 tests, 6 on Windows, 1 on Linux, on 4 versions of R. I can maybe make more tests on Linux on my other computers but if it's not a single issue it will be very very hard to reproduce it and then, to solve it. 

Thank you for your tests. I will consider this bug fixed because we cannot reproduce it. It seems to be specific to something else than pure code.

ptompalski, 2016-03-23:
Agree. If it happens again, then we can investigate it more.

closed 2016-03-23 by Jean-Romain