r-lidar/lidR #5: Catalog LasLib error

pdbentley, 2016-08-22:
Hi Jean-Romain, 

thank you for putting this package together. i was about to start in this space then discovered your work and i like the direction it is heading in. 

i have a couple of lidar based projects at the moment so can help road test this package. 

the issue: 

with Catalog you moved to laslib in recent versions. i find i cannot read more than ~500 las files before LASlib gets an internal error.  From what i can see it is about the number of files. the original version of lidR 1.0.0 did not have this issue. 

thoughts?

Paul Bentley

Jean-Romain, 2016-08-22:
Hi Paul

Could you please send me the code that create the bug and the error message you got in return.

The version 1.0.0 used its own function in pure R to read las files. However I wasn't able to be sure that it worked properly for every files. Furthermore I wasn't able to write las files for technical raison due to R itself (too high level language). This is why I moved to C++. I linked the LASlib from Martin Isenburg. It enables me to rely on a more advanced tools and provides me the ability to deal with laz files. If the problem really comes from LASlib I can try to track the bug but I can't be sure to succeed. Have you ever try to deal with 500 files with LAStools software suit?

Thank you for reporting the bug.

JR

pdbentley, 2016-08-22:
Try this. I am using RStudio in Windows 8 at the moment btw.  After i try to run Catalog, this message is repeated on the console. 

_ERROR: cannot open lasreaderlas with file name 'D:\2016_EdenManagementArea\LAS\Area_1234\752000_5861000.las'
ERROR: cannot open fiError: LASlib internal error. See message above.
le 'D:\2016_EdenManagementArea\LAS\Area_1234\752000_5862000.las'
ERROR: cannot open lasreaderlas with file name 'D:\2016_EdenManagementArea\LAS\Area_1234\752000_5862000.las'
ERROR: cannot open file 'D:\2016_EdenManagementArea\LAS\Area_1234\752000_5863000.las'
ERROR: cannot open lasreaderlas with file name 'D:\2016_EdenManagementArea\LAS\Area_1234\752000_5863000.las'
Error: LASlib internal error. See message above.
ERROR: cannot open file 'D:\2016_EdenManagementArea\LAS\Area_1234\752000_5864000.las'
ERROR: cannot open lasreaderlas with file name 'D:\2016_EdenManagementArea\LAS\Area_1234\752000_5864000.las'
Error: LASlib internal error. See message above._

However, when i go to enter another command into the RStudio console i get this message:

_Error in gzfile(file, "rb") : cannot open the connection
In addition: Warning message:
In gzfile(file, "rb") :
  cannot open compressed file 'C:/Users/Paul/Documents/Projects/2016_PHD_THB/PHD_THB_201608_EDEN/packrat/lib/x86_64-w64-mingw32/3.3.0/Rcpp/Meta/package.rds', probable reason 'Too many open files'_

Maybe its RStudio??

Given how LAStools is used i don't think an error like this would be there. i suspect it is somewhere in implementation. i have played with LASTools but not that many files; i have mostly been using FUSION.

pdbentley, 2016-08-22:
here is the code is used (

_library(rgeos)
library(sp)
library(rgdal)
library(lidR)
library(rLiDAR)

#get summary
lashome <- "D:\\2016_EdenManagementArea\\LAS\\Area_05\\"
lascat<- Catalog(lashome,recursive = TRUE)_
 
and session info...

R version 3.3.0 (2016-05-03)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows >= 8 x64 (build 9200)

locale:
[1] LC_COLLATE=English_United States.1252  LC_CTYPE=English_United States.1252   
[3] LC_MONETARY=English_United States.1252 LC_NUMERIC=C                          
[5] LC_TIME=English_United States.1252    

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] rLiDAR_0.1   lidR_1.1.1   dtplyr_0.0.1 magrittr_1.5 rgdal_1.1-10 sp_1.2-3     rgeos_0.3-19

loaded via a namespace (and not attached):
 [1] Rcpp_0.12.6      formatR_1.4      plyr_1.8.4       bitops_1.0-6     tools_3.3.0      magic_1.5-6     
 [7] rpart_4.1-10     goftest_1.0-3    digest_0.6.10    packrat_0.4.7-1  nlme_3.1-127     evaluate_0.9    
[13] tibble_1.1       lattice_0.20-33  mgcv_1.8-12      Matrix_1.2-6     DBI_0.4-1        yaml_2.1.13     
[19] parallel_3.3.0   spam_1.3-0       dplyr_0.5.0      stringr_1.0.0    knitr_1.14       raster_2.5-8    
[25] fields_8.4-1     maps_3.1.1       grid_3.3.0       data.table_1.9.6 R6_2.1.2         rgl_0.95.1441   
[31] rmarkdown_1.0    polyclip_1.5-6   tensor_1.5       reshape2_1.4.1   deldir_0.1-12    htmltools_0.3.5 
[37] abind_1.4-5      assertthat_0.1   spatstat_1.46-1  stringi_1.1.1    geometry_0.3-6   chron_2.3-47

Jean-Romain, 2016-08-22:
It looks to be a internal error of LASlib when reading `752000_5861000.las` an 5 other files. LASlib is not very informative on the error. From my tests, it can returns this error at least in two cases:

- When the file is corrupted or not valid regarding LAS specifications
- When the rights on the file do not allow to read it.

What happens if you try:

    readLAS("D:\2016_EdenManagementArea\LAS\Area_1234\752000_5861000.las")

if it does not work, what happens if you execute the command `lasinfo` (from lastools or from [liblas](http://www.liblas.org/) or maybe there is an equivalent one in FUSION) on this file.

Jean-Romain, 2016-08-22:
Ok I checked the code, I'm write I didn't close the connection into the `readLASheader` function. So basically you opened 500 connections without closing them and everything crash. I will commit a patch today. It might solves the issue

Jean-Romain, 2016-08-22:
I updated a new version (1.1.2) in the development branch. Install this version and tell me is the problem is solved.

    devtools::install_github("Jean-Romain/lidR", ref="develop")

pdbentley, 2016-08-23:
that does the trick.  thanks.

closed 2016-08-23 by pdbentley