r-lidar/lidR #6: catalog reading .lasx files

pdbentley, 2016-08-23:
files = list.files(folder, full.names = T, pattern = "\\.las|.laz$", ...)

pdbentley, 2016-08-23:
Hey JR, 

ill add a couple of other small things i've come across. 

firstly, thanks for updating the list.files. it works, however i also have .lasx files (its an ESRI stats file). i had a quick look but could find a simple way to update the pattern recognition bit. i use grep alot, but not sure if it works here.

Jean-Romain, 2016-08-23:
Hi, 

Not sure to understand the question. You mean that lasx files are recognize in the following pattern `\\.las|.laz$` and you want the pattern to do not take into account these files keeping only las and laz?

If yes it easy I'll do that.

Jean-Romain, 2016-08-23:
Done is `develop` branch.

pdbentley, 2016-08-25:
great.  thanks. that will stop it falling over from pesky extension names.

closed 2016-08-25 by pdbentley