r-lidar/lidR #7: gapFractionProfile for some dz

pdbentley, 2016-08-23:
i like this function. 

it works for for some dz, but for others i get this error. 
  
  Error in hist.default(z, breaks = bk, plot = F) : 
  some 'x' not counted; maybe 'breaks' do not span range of 'x'

for example the function would work for dz=1.9 and dz=2.1, but not for dz=2. 

looking at code this error is coming from the hist() function. im not sure what the trigger is.

Jean-Romain, 2016-08-23:
The problem effectively comes from `hist` it is a known bug (but I forgot it as I didn't use this function since a while).

If your `Z` elevations range between 0 and 20 (it is always rounded) for example, the `dz` must be a multiple of the range size (here 20) to make a integer number of bin. So 1.9 does not make sense. 0.5, 1 and 2 work.

I don't know what is the best behavior. I can change the behavior to enable to create any bin size. It is probably better you're right.

Jean-Romain, 2016-08-23:
Done in `develop` branch

Now if max Z is 44.6 for example and you ask for 1.3 m bins the bins will be [0, 1.3[, ]1.3, 2.6[, .... ]44.2, 45.5] without bug in the range of created bins.

closed 2016-08-25 by Jean-Romain