r-lidar/lidR #13: Error while trying to load lidar files with RGB values

ptompalski, 2016-08-30:
When trying to read a .las file that contains RGB attributes the following error occurs:

`> a <- readLAS("data/a_3410263_20_20.las")`
`Error in grDevices::rgb(R/255, G/255, B/255) : `
`  color intensity 54.2118, not in [0,1]`

Jean-Romain, 2016-08-30:
Crap! I knew it ! RGB is a problematic field. The official LAS specification say that 16 bits (unsigned short) are given for each color. It means that colors range between 0 and 65535. But some files do not respect standard and they do no normalize colors to 16 bits (keeping the 8 bits value ranging from 0 to 255).

This is what was done in the file I tested. Your file is a proper LAS file following the standard. But my code do not respect the standard because I tested it in file which does not.

If you do not need the RGB field, do not load it using `RGB = FALSE` (since version 1.10). Else I will correct it soon.

Jean-Romain, 2016-08-30:
This is corrected in version 1.2.0 in the `devel` branch. The version 1.2.0 introduce many many changes and incompatibilities with previous versions (function names changed). Please read the [changelog](https://github.com/Jean-Romain/lidR/tree/devel#changelog-v120) carefully.

If you are still in version < 1.0 read carefully how to install the package now (and read the whole changelog). Note that `devtools:install_github` has a param `ref`. Set it to `ref="devel"`to install from the development branch rather than `master` branch.

Thanks for reporting the bug.

ptompalski, 2016-08-30:
Thanks JR!

Piotr Tompalski
Postdoctoral Research Fellow
Integrated Remote Sensing Studio <http://irsslab.forestry.ubc.ca/> | University
of British Columbia <http://www.ubc.ca>
Research Gate <https://www.researchgate.net/profile/Piotr_Tompalski> |
@PiotrTompalski <https://twitter.com/PiotrTompalski>

commit ff04bfb8 (2016-08-30): Correct RGB computation for 16 bits data (close #13)
closed 2016-09-07 by Jean-Romain