r-lidar/lidR #16: normalize - possible interp issue??

pdbentley, 2016-09-08:
Hi Jean-Romain, 

I am back in front of a computer and have been road testing your package again. i am looking at overlapping data sets, so i have one from 2008 and one from 2012. Normalize works for the 2012, but not the 2008. below is the input class 'LAS' file i am using - both for the same plot. i tried looking into the error, but not sure why one dataset creates the issue, the other doesn't. the things which stands out is the LAS version, but doesn't directly match the error. 

Thanks in advance. 

**this is the error** (hence why i say interp)

Error in if (drx/dry > 10000 || drx/dry < 1e-04) stop("scales of x and y are too dissimilar") : 
  missing value where TRUE/FALSE needed
In addition: Warning messages:
1: In min(data$X) : no non-missing arguments to min; returning Inf
2: In min(data$Y) : no non-missing arguments to min; returning Inf
3: In min(data$Z) : no non-missing arguments to min; returning Inf
4: In max(data$X) : no non-missing arguments to max; returning -Inf
5: In max(data$Y) : no non-missing arguments to max; returning -Inf
6: In max(data$Z) : no non-missing arguments to max; returning -Inf
7: In min(x) : no non-missing arguments to min; returning Inf
8: In max(x) : no non-missing arguments to max; returning -Inf
9: In min(x) : no non-missing arguments to min; returning Inf
10: In max(x) : no non-missing arguments to max; returning -Inf

**This data set does not work.**

An object of class "LAS"
Slot "data":
            X       Y      Z Intensity ReturnNumber Classification ScanAngle  gpstime pulseID flightlineID
  1: 350459.3 5846009 556.60        31            1             10        -6 19564.29  229075            2
  2: 350459.1 5846008 553.12         4            2             10        -6 19564.29  229075            2
  3: 350458.4 5846008 554.98        29            1             10        -6 19564.29  229076            2
  4: 350458.1 5846007 550.17         1            2             10        -6 19564.29  229076            2
  5: 350457.9 5846008 558.94         4            1             10        -6 19564.29  229077            2
 ---                                                                                                      
649: 350442.9 5846020 554.24         8            2             10        -6 19564.57  238308            2
650: 350442.3 5846018 543.86         1            3             10        -6 19564.57  238308            2
651: 350443.8 5846020 542.79         6            2             10        -5 19564.57  238310            2
652: 350444.6 5846021 542.86         0            3             10        -5 19564.57  238311            2
653: 350445.4 5846022 543.66         0            4             10        -5 19564.57  238312            2

Slot "area":
[1] 304.8

Slot "pointDensity":
[1] 2.14

Slot "pulseDensity":
[1] 1.12

Slot "header":
$`File Signature`
[1] "LASF"

$`File Source ID`
[1] 0

$`Global Encoding`
[1] 0

$`Project ID - GUID`
[1] 0

$`Version Major`
[1] 01

$`Version Minor`
[1] 00

$`System Identifier`
[1] ""

$`Generating Software`
[1] "TerraScan"

$`File Creation Day of Year`
[1] 0

$`File Creation Year`
[1] 0

$`Header Size`
[1] 227

$`Offset to point data`
[1] 229

$`Number of variable length records`
[1] 0

$`Point Data Format ID`
[1] 01

$`Point Data Record Length`
[1] 28

$`Number of point records`
[1] 6237513

$`Number of 1st return`
[1] 3601894

$`Number of 2nd return`
[1] 1811872

$`Number of 3rd return`
[1] 709500

$`Number of 4th return`
[1] 114247

$`Number of 5th return`
[1] 0

$`X scale factor`
[1] 0.01

$`Y scale factor`
[1] 0.01

$`Z scale factor`
[1] 0.01

$`X offset`
[1] 0

$`Y offset`
[1] 0

$`Z offset`
[1] 0

$`Max X`
[1] 350460.9

$`Min X`
[1] 350441.2

$`Max Y`
[1] 5846024

$`Min Y`
[1] 5846004

$`Max Z`
[1] 568.68

$`Min Z`
[1] 529.89

**This dataset does work**

An object of class "LAS"
Slot "data":
             X       Y      Z Intensity ReturnNumber Classification ScanAngle     R     G     B  gpstime pulseID flightlineID   color
   1: 350460.2 5846018 540.06        23            2              5       -18 14336 62208 10240 13386014 1543063            1 #38F228
   2: 350459.7 5846016 537.55        34            2              4       -18 14336 62208 10240 13386014 1543064            1 #38F228
   3: 350459.7 5846016 537.35        36            2              4       -18 14336 62208 10240 13386014 1543065            1 #38F228
   4: 350459.7 5846015 537.88        18            4              5       -18 14336 62208 10240 13386014 1543066            1 #38F228
   5: 350460.5 5846016 543.02         4            3              5       -18 14336 62208 10240 13386014 1543067            1 #38F228
  ---                                                                                                                                
2687: 350441.1 5846014 562.64        41            1              5       -18 14336 62208 10240 13386015 1704687            1 #38F228
2688: 350441.2 5846014 562.67        52            1              5       -18 14336 62208 10240 13386015 1704688            1 #38F228
2689: 350441.1 5846014 561.88        18            1              5       -18 14336 62208 10240 13386015 1704689            1 #38F228
2690: 350441.0 5846015 560.34        33            1              5       -18 14336 62208 10240 13386015 1704691            1 #38F228
2691: 350441.1 5846015 560.46        60            1              5       -18 14336 62208 10240 13386015 1704692            1 #38F228

Slot "area":
[1] 305.5

Slot "pointDensity":
[1] 8.81

Slot "pulseDensity":
[1] 4.65

Slot "header":
$`File Signature`
[1] "LASF"

$`File Source ID`
[1] 0

$`Global Encoding`
[1] 1

$`Project ID - GUID`
[1] 0

$`Version Major`
[1] 01

$`Version Minor`
[1] 02

$`System Identifier`
[1] "LAStools (c) by rapidlasso GmbH"

$`Generating Software`
[1] "las2las (version 160606)"

$`File Creation Day of Year`
[1] 103

$`File Creation Year`
[1] 2012

$`Header Size`
[1] 227

$`Offset to point data`
[1] 229

$`Number of variable length records`
[1] 0

$`Point Data Format ID`
[1] 03

$`Point Data Record Length`
[1] 34

$`Number of point records`
[1] 4947397

$`Number of 1st return`
[1] 2992076

$`Number of 2nd return`
[1] 1288598

$`Number of 3rd return`
[1] 506869

$`Number of 4th return`
[1] 159854

$`Number of 5th return`
[1] 0

$`X scale factor`
[1] 0.01

$`Y scale factor`
[1] 0.01

$`Z scale factor`
[1] 0.01

$`X offset`
[1] 0

$`Y offset`
[1] 0

$`Z offset`
[1] 0

$`Max X`
[1] 350460.6

$`Min X`
[1] 350441

$`Max Y`
[1] 5846024

$`Min Y`
[1] 5846004

$`Max Z`
[1] 568.44

$`Min Z`
[1] 530.3

Jean-Romain, 2016-09-08:
Hi, the DTM normalization is an experimental feature. I tested it in few dataset but I don't have a lot data to test it. I published the feature expecting somebody to find a bug. So thank you for testing it.

I approximately know where it can be bugged in my code. But the error you copied is not from my code. The warning messages could come from my code but I'm not sure.

First, are you using your own DTM from an external file or did you compute it with the `grid_terrain()` function? It seems that it is related to scale. Could you send me a summary of the content of your DTM.

pdbentley, 2016-09-08:
i didn't use a dtm, i used the internal  grid_terrain. if i run
grid_terrain by itself i get the same error message.

i am using a circular plot. the one i sent earlier was a 10m radius. i just
tried larger circular plot sizes up to 50m with the same error.

i get the impression there is something in the 2008.las file which
generates the error as the 2012.las file is fine....

Jean-Romain, 2016-09-08:
First, as explained in the doc, `grid_terrain` is a only an interpolation of ground returns. It is probably not as powerful as the DTM functions provided in FUSION for example. It's nice to have a quick look to normalized data but my advice, if you want to properly compare two datasets, is to use more advanced functions which are known to be good methods for terrain model.

Concerning your problem, which is not related to the quality of the algorithm, I don't have a clue. Could you write your circular plot in a laz file (`writeLAS`) and send it to me with the minimal R code required to reproduce the bug.

Considering your anwser, the bug comes from `grid_terrain` not `normalize`. It is the first clue.

Jean-Romain, 2016-09-08:
I think I got it.

Ground return are classified `2` considering the [LAS specification](http://www.asprs.org/wp-content/uploads/2010/12/LAS_1_4_r13.pdf). The classification number in your first dataset look to be `5` for each points (hight vegetation). Is it possible that you do not have point classify as "ground".

Try to use function `getGround`. If it returns nothing the problem is the following one. This hypotheses matches with the warning messages.

Jean-Romain, 2016-09-08:
I succeeded in reproducing the bug. Your las file has no point classified as ground. I can't do anything for you within my package.

Now if you run a filter that return nothing (in you case you runned `getGround` via `grid_terrain`) it stop the script and send an explicit error. For example  `Error : No point found with the condition: Classification == 2`.

pdbentley, 2016-09-08:
that makes sense. i am working on two version - one combined and one split.
So grid_terrain explicitly looks for the Classification to calculate the
ground.

Its a good error to add as there will likely be other data sets with a
missing classification.

Thanks.

closed 2016-09-09 by Jean-Romain