r-lidar/lidR #17: readLAS

pdbentley, 2016-09-08:
readLAS already exists in the package 'Rlidar' - errors occur with both packages loaded and using this function. 

solution to allow the other single input from the other package? or rename?

pdbentley, 2016-09-08:
rLiDAR is the package

Jean-Romain, 2016-09-08:
I will not rename. I know that `rlidar` have a `readLAS` function. Our two function are very very different. Mine is much more powerfull but if you want to use functions from `rlidar` the formats are not compatible. What I can do is to add a function `as.rlidar` do convert data into a compatible form for this package. If you want to use a specific `readLAS` function you can call it with the namespace.

- `rlidar::readLAS()`
- `lidR::readLAS()`

pdbentley, 2016-09-08:
fair point.

i am playing around with other functions in rLiDAR at the moment for tree
identification; hence why they were both in action. with both packages open
confusion was created. i can be more specific. thanks.

closed 2016-09-08 by Jean-Romain