r-lidar/lidR #28: R crash while writing files with RGB

wijnland, 2016-11-07:
When trying to write a las file (writeLAS) my R-session crashes. The problem only seems to occur only when RGB are loaded. 
lidR: 1.2.0   (ref="devel')
R 3.3.2 / Platform: x86_64-w64-mingw32/x64 (64-bit). 

the problem seems to occur erratic, first save after rebooting works sometimes, but the colors in the created file look wrong (r/g/b in wrong order in part of the file)
I am currently working with Pix4D generated las files that are photogrammetry based.

--- crash details from windows --
Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	rsession.exe
  Application Version:	1.0.44.0
  Application Timestamp:	5806b90d
  Fault Module Name:	lidR.dll
  Fault Module Version:	0.0.0.0
  Fault Module Timestamp:	5820df55
  Exception Code:	c0000005
  Exception Offset:	000000000006979f
  OS Version:	6.1.7601.2.1.0.256.4
------

Jean-Romain, 2016-11-07:
Hi,

Could you send me a small las file and a toy example of code to reproduce the bug. I do not have las file with RGB fields. Furthermore I do not use the `writeLAS` function for my own purpose I wrote it because somebody asked it to me. So I need to make some test. I don't have a clue right now.

If the session crashes, it means that there is a bad memory allocation somewhere in C++ code.

wijnland, 2016-11-08:
I can, although apparently not here (both my .las and .zip are not accepted). file on its way by email.

with the much smaller file i clipped\decimated for sending R does not crash, but the colors are still wrong on saving.

the simplest code to reproduce would be:
l <- readLAS("P4Dlas_clip.las")
writeLAS(l, "outfile.las")

when reading "outfile.las" back in and plotting with color="color" it shows they are all wrong (as when examined in other tool)

wijnland, 2016-11-08:
[P4Dlas_clip.gz](https://github.com/Jean-Romain/lidR/files/576989/P4Dlas_clip.gz)
so i guess emailing did not work, but .gz does!

Jean-Romain, 2016-11-08:
Got it. I had no crash problem when I wrote the file. But the color are wrong when I load it back. I'm investigating.

Jean-Romain, 2016-11-08:
Problem solved in commit  e1ae09a in `devel` branch. I forgot a variable in the source code. The las file was badly wittten e.g. green field was in gpstime and other switches were made. 

It solves your problem of color when reading back the file. I don't know were does the crashes come from but I suspect it comes from bad type casting due this error. So I hope I solved the two bug.

wijnland, 2016-11-08:
yes. Thanks. This solves the issue!

commit e3a13eee (2016-11-08): Fix variable ommision introducing bug (close #28)
closed 2016-11-08 by Jean-Romain