r-lidar/lidR #29: classify_from_shapefile ignores multipart

wijnland, 2016-11-15:
Hi, I have been trying to use the classfy_from_shapefile function and got unexpected results. after some digging i found that the function seems to only include the first slot of any PolygonsDataFrame.
the behavior seems the same whether it is a multipart polygon, or a shapefile with multiple polygons.

\on a side note:
are there very specific reasons why the grid_metrics and related function produce their own data type. it may be desirable to create grids that use a use a compatible format with the 'sp' or 'raster' libraries. for seamless use with many other functions there.

wijnland, 2016-11-16:
also holes in polygons seem to be ignored at this point.

Jean-Romain, 2016-11-16:
Hi. It's possible that this function would not be able to manage every case. I designed this function based on what I know about shapefiles and their representation in R i.e. not a lot of things. It works when the polygons are simple plain and not fragmented shapes.

> i found that the function seems to only include the first slot of any PolygonsDataFrame

Send me the shapefile. I will look how does it look like. I think I can easily manage multipart polygons but I'm not sure to be able to quickly manage polygon with holes.

> are there very specific reasons why the grid_metrics and related function produce their own data type.

There are so many reasons why I use a database structure rather than an array structure. Let me list some of them.

1. If you want to easily compute a predictive model from your metrics what you need is a data base structure.`mymetrics$prediction = a*mymetrics$metric1 + b*mymetrics$metric2^c`
2. If you use the `split_flightline' option. In this case you can have some pixel twice. This cannot be represented in an array structure.
3. If you compute several metrics you need a multilayered array which is a very inefficient data structure. The database structure is good.
4. The way I compute the grid_metric is very efficient and super fast using the power of the data.table package. And the natural output structure is a data.table.

The most of the time what is needed is a database structure not an array structure. If you really need a spatial structure of the data such as a raster from raster package you can transform the data.

```r
library(raster)
mymetric %>% as.matrix %>% raster
```

ps: I don't know how your learn about the existence of this package but be aware about the fact that lidR this is a single guy project designed for myself. I provide it for free but I don't know everything so I can make mistake. Moreover I didn't designed it to any possible need ; only for my own purposes + some features people asked to me. So do not hesitate to tell me if something sounds wrong in your opinion. You can report bug, I'm grateful, but you can also argue that some features are not well designed. If your arguments sounds good I will improve the functions in your way.

Jean-Romain, 2016-11-16:
I updated the code in the devel branch to deal with multi-part polygons. I didn't know that such feature was possible that is why I didn't manage this case and, as you said,  I only included the first slot of any PolygonsDataFrame.

I don't have shapefiles with multi-part polygons. I tried to create a dummy file but my QGIS has problem and crashed. So I didn't actually test the feature with real lidar data. I only checked if I didn't breaks the old behavior. Could you confirm it has the expected behavior with multi-part polygon?

I considered that a polygon could be composed of 1 polygon or more. But I didn't manage the case when sub-polygons are themselve composed of sub-polygons. I mean second order multi-part polygons. I don't know if it could exists. It yes I had to write a recursive version... I hope it is not possible.

About the hole in the polygons I don't know how to draw such thing. S I cannot check how it is represented in a shapefile. So I didn't manage this case. Do you have a shapefile with holes in polygons?

wijnland, 2016-11-16:
attached a shape with two parts and one with hole. hope this helps. 
[polygon.tar.gz](https://github.com/Jean-Romain/lidR/files/595918/polygon.tar.gz)

thanks for all the quick responses and fixes, really great. 
I understand the format of the grid metrics now, that makes a lot of sense. I did not catch on that they were data tables first which gave me some issues casting them into another format. my thought about it it was not really to make them rasters, but rather to have the table compatible with something alike the sp::SpatialPixels or sp::SpatialGrid classes, which inherit projections etc. and are easy to convert rasters if needed. I am unsure how they relate to data tables, and honestly the conversion to raster you show works fine.
Piotr at my former lab (UBC-IRSS) pointed me towards lidR. great work!

Jean-Romain, 2016-11-16:
I added the feature to take account of holes in polygons. I manage to make my own shapfile with holes and multi-part polygons within my lidar data. It works in my tests but I cannot state it's bug free yet. Test it and let me know if you have a problem.

> my thought about it it was not really to make them rasters, but rather to have the table compatible with something alike the sp::SpatialPixels or sp::SpatialGrid classes which inherit projections etc. and are easy to convert rasters if needed.

I'm not comfortable with spatial data, projection, `rgdal`, `raster`, etcetera. Especially I alway get mad with projection when I have to create some shapefiles (today for example). This is not my field of knowledge. That is why I skipped all these features. But if you tell me how to properly create such data structure I can provide functions to easily make some type conversions. However I will never use these structures natively because they aren't appropriate to deal with large amount of data in the way I want.

wijnland, 2016-11-17:
the holes work for me, no further issues so far.

understandable, the whole spatial stuff is a whole new can of worms. and although i deal with it a a bit dont think i know of all of the ins and outs. I think the main thing of the classes is that they have a slot of the the projection (CRS), and some specific slots for coordinates and bounding box etc which can be used by specific spatial methods. since las data is very spatial, it would be great to follow those standards. It would be great to have at least the projection included by the LAS (read/write) functions (currently that info is dropped), the other info is available now even if it is not directly compatible.
I would be interested in helping out on some of this, lidR fills a great need (for me at least...). but i must admit that the programming is a bit over my head for most parts

Jean-Romain, 2016-11-17:
The projection is not a required field in the LAS specifications. I can only add something to read the additional data which could be included in the file. Sometime the provider chose to add the projection in such field but not necessarily. And your are right I dropped this optional data.

wijnland, 2016-11-17:
ah, I sort of assumed it would be in each lasfile. It does kind of makes sense to not include it if is in the additional fields as they would not be uniformly formatted.
I will just keep messing around, and will drop in here if i find more unexpected behaviors. Thanks for implementing the multipart polygons and holes.

closed 2016-11-17 by Jean-Romain

Jean-Romain, 2016-11-29:
Hi, 

I don't know if you will receive an email from a closed issue but there is now a function `as.raster` to cast the  `grid_metric` output into a proper `RasterLayer` object. The function `as.matrix` has been removed.