r-lidar/lidR #39: lidR bug

japolo, 2017-01-12:
Hi,
I am trying lidR for the first time. This is also the first time I've ever reported a bug and I'm not very familiar with GitHub, so sorry if I'm missing something. 

>library(lidR)
>lascat <- catalog("c:/users/p/documents/gis")
>palas <- readLAS(lascat)
>cano1 <-grid_canopy(palas)
>terr1 <-grid_terrain(palas, method="kriging")

[using universal kriging]
Error in predict.gstat(g, newdata = newdata, block = block, nsim = nsim,  : 
  <NULL> message: indicating a software bug, please report

>sessionInfo()

R version 3.3.1 (2016-06-21)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows 7 x64 (build 7601) Service Pack 1

locale:
[1] LC_COLLATE=English_United States.1252 
[2] LC_CTYPE=English_United States.1252   
[3] LC_MONETARY=English_United States.1252
[4] LC_NUMERIC=C                          
[5] LC_TIME=English_United States.1252    

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] lidR_1.0.2   magrittr_1.5

loaded via a namespace (and not attached):
 [1] Rcpp_0.12.7      lattice_0.20-34  intervals_0.15.1 FNN_1.1         
 [5] spacetime_1.2-0  zoo_1.7-13       gstat_1.1-3      chron_2.3-47    
 [9] grid_3.3.1       lazyeval_0.2.0   data.table_1.9.6 raster_2.5-8    
[13] sp_1.2-3         xts_0.9-7        tools_3.3.1      rlas_1.0.3

Jean-Romain, 2017-01-12:
Hi

The error occured in `gstat` package. Does it work with the file provided in the examples?(Topography.laz)
- If yes please try to send me a small file that enable to reproduce the bug
- If no... I don't know.  We will see later

Another point, you used `readLAS` with a `Catalog`. Why not, the function has been designed for that. But how many file did you read? I mean maybe you tried to run the example on a huge amount of data. I worked a lot to optimize many many things but processing a whole Catalog is maybe to much.

japolo, 2017-01-12:
Jean-Romain,
Thank you for replying. 

I tried again with Topography.laz. 

> topo1 <- readLAS("C:/Users/p/Documents/GIS/temp/Topography.laz")
> terr1 <- grid_terrain(topo1, method="kriging", model= gstat::vgm("Gau"))

Error in if (any(model$range < 0)) { : 
  missing value where TRUE/FALSE needed
> traceback()

9: load.variogram.model(object$model[[name]], c(i - 1, i - 1), max_dist = max_dist)
8: predict.gstat(g, newdata = newdata, block = block, nsim = nsim, 
       indicators = indicators, na.action = na.action, debug.level = debug.level)
7: predict(g, newdata = newdata, block = block, nsim = nsim, indicators = indicators, 
       na.action = na.action, debug.level = debug.level)
6: .local(formula, locations, ...)
5: gstat::krige(Z ~ X + Y, location = ~X + Y, data = ground, newdata = coord, 
       model)
4: gstat::krige(Z ~ X + Y, location = ~X + Y, data = ground, newdata = coord, 
       model)
3: terrain_kriging(ground, coord, model)
2: lasterrain(.las, grid, ...)
1: grid_terrain(topo1, method = "kriging", model = gstat::vgm("Gau"))

To reply to your second point, the Catalog that I was using at first had only 2 .las files in it, it wasn't too big. I wasn't sure of the proper flow for getting from .las to creating canopy and DEM files, so I thought it should go .las > Catalog > readLAS > grid_canopy & grid_terrain. I'll try to read more. 

John

Jean-Romain, 2017-01-12:
```
terr1 <- grid_terrain(topo1, method="kriging", model= gstat::vgm("Gau"))
```

Try it without `model` using the default parameters. I'm not sure your variogram is correct. The default parameters are expected to work.

> To reply to your second point, the Catalog that I was using at first had only 2 .las files in it, it wasn't too big.

That's good

japolo, 2017-01-12:
On 1/12/2017 11:49 AM, Jean-Romain Roussel wrote:
> |terr1 <- grid_terrain(topo1, method="kriging", model= gstat::vgm("Gau")) |
>
> Try it without |model| using the default parameters. I'm not sure your 
> variogram is correct. The default parameters are expected to work.
>
If I use sample data, Topography, and the default parameters, it works. 
It still won't work with my data though. Here are the two sets of data:

 > summary(topo1) #Topography.laz
class        : LAS
memory       : 2.8 Mb
extent       : 684800 , 685150 , 5018000 , 5018300 (xmin, xmax, ymin, ymax)
area         : 104934.6 m^2
points       : 52146 points
pulses       : 1 pulses
point density: 0.5 points/m^2
pulse density: 0 pulses/m^2
fields       : 10
field names  : X Y Z gpstime Intensity ReturnNumber NumberOfReturns 
Classification ScanAngle pulseID

 > summary(palas) #my catalog of two .las files
class        : LAS
memory       : 477.5 Mb
extent       : 719350.2 , 725202.6 , 3761248 , 3764837 (xmin, xmax, 
ymin, ymax)
area         : 19934430 m^2
points       : 8940868 points
pulses       : 7568460 pulses
point density: 0.45 points/m^2
pulse density: 0.38 pulses/m^2
fields       : 10
field names  : X Y Z gpstime Intensity ReturnNumber NumberOfReturns 
Classification ScanAngle pulseID

Maybe my file is too big? I don't know what the upper limit is for size. 
My laptop only has 8 GB of RAM, but I thought it would be enough.

John

Jean-Romain, 2017-01-12:
First conclusion, there is neither a bug in my code nor a problem with `gstat` since it works on your computer with examples

So it comes from your data. My first guess is the amount of data. I mean, with 2.8 Mb of data the kriging takes several minutes on my computer. So do not expect to perform that method with 470 Mb. But this explanation is not satisfactory. It should work even if it requires a week of computation.

I'll try to run that code with more data (actually I never try). Anyway, bug or not bug you will never be able to run that method on a so big dataset. The algorithm it too slow.

japolo, 2017-01-12:
On 1/12/2017 2:45 PM, Jean-Romain Roussel wrote:
> First conclusion, there is neither a bug in my code nor a problem with
> |gstat| since it works on your computer with examples
>
> So it comes from your data. My first guess is the amount of data. I [...]
Okay, thank you. I guess it's the kriging that's the problem. I can use 
other LiDAR software, like LASTools, with these files. It takes a minute 
or two to build a D.E.M., but it can be done. I would rather use R, 
since I do a lot of other stuff in R and it would be great to take care 
of LiDAR processing there too. I'm pretty sure that the other software 
don't use kriging.

Jean-Romain, 2017-01-12:
DTM is my stronger issue with `lidR`. I proposed some algorithms but I'm not able to process big dataset yet. I believe that open source algorithm are important for such task but I'm currently not enough efficient. You are better to use LAStools to process big datasets. The problem is that it relies on a close source algorithm.

Jean-Romain, 2017-01-13:
Hi again,

I will release a new version on github tomorrow with is both a krigging method with a knn approch. The previous method used all the ground points. But actually only the k-nearest neighbourg are needed. The computation is incredibly much faster

Jean-Romain, 2017-01-13:
Try the new version `lidR v1.1.0`. 

```r
devtools::install_github("Jean-Romain/lidR", ref="devel")
```

closed 2017-01-15 by Jean-Romain

japolo, 2017-01-18:
Hi Jean-Romain,

> devtools::install_github("Jean-Romain/lidR", ref="devel")
Downloading GitHub repo Jean-Romain/lidR@devel
from URL https://api.github.com/repos/Jean-Romain/lidR/zipball/devel
Installing lidR
...
** testing if installed package can be loaded
*** arch - i386
*** arch - x64
* DONE (lidR)

> library(lidR)
Loading required package: magrittr
lidR 1.1.0.9000. For help, for the most up-to-date version or to report a bug, visit https://github.com/Jean-Romain/lidR

> readLAS("c:/users/p/documents/gis/33096H5_NW_A_Reprj26914.las")
Error in rlas::readlasdata(file, Intensity, ReturnNumber, NumberOfReturns,  : 
  unused argument (filter)

I can't get my .las file in to try to test the new grid_terrain().

Jean-Romain, 2017-01-18:
Update  also my rlas package

```
devtools::install_github("Jean-Romain/rlas")
```

japolo, 2017-01-19:
I let the function run for more than 10 hours and then I cut it off. I guess you could say it was working, but it takes too long. 

> grid_terrain(testlas, method = "kriging", k = 10)
[using universal kriging]

There were 50 or more warnings (use warnings() to see the first 50)

> warnings()
Warning messages:
1: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723755,3.7614e+006,0]: skipping...
2: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723756,3.7614e+006,0]: skipping...
3: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723757,3.7614e+006,0]: skipping...
4: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723758,3.7614e+006,0]: skipping...
5: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723759,3.7614e+006,0]: skipping...
6: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723755,3.7614e+006,0]: skipping...
7: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723757,3.7614e+006,0]: skipping...
8: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723758,3.7614e+006,0]: skipping...
9: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723759,3.7614e+006,0]: skipping...
10: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723760,3.7614e+006,0]: skipping...
11: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [725065,3.7614e+006,0]: skipping...
12: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [725066,3.7614e+006,0]: skipping...
13: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723754,3.7614e+006,0]: skipping...
14: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723756,3.7614e+006,0]: skipping...
15: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723757,3.7614e+006,0]: skipping...
16: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723758,3.7614e+006,0]: skipping...
17: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723759,3.7614e+006,0]: skipping...
18: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723760,3.7614e+006,0]: skipping...
19: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723754,3.7614e+006,0]: skipping...
20: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723755,3.7614e+006,0]: skipping...
21: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723756,3.7614e+006,0]: skipping...
22: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723758,3.7614e+006,0]: skipping...
23: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723759,3.7614e+006,0]: skipping...
24: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723761,3.7614e+006,0]: skipping...
25: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723753,3.7614e+006,0]: skipping...
26: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723754,3.7614e+006,0]: skipping...
27: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723755,3.7614e+006,0]: skipping...
28: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723756,3.7614e+006,0]: skipping...
29: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723761,3.7614e+006,0]: skipping...
30: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723753,3.7614e+006,0]: skipping...
31: In predict.gstat(g, newdata = newdata, block = block,  ... :
[... 14 lines cut ...]
  Covariance matrix singular at location [723755,3.7614e+006,0]: skipping...
39: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723758,3.7614e+006,0]: skipping...
40: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723759,3.7614e+006,0]: skipping...
41: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723760,3.7614e+006,0]: skipping...
42: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723755,3.7614e+006,0]: skipping...
43: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [723756,3.7614e+006,0]: skipping...
[... cut ...]
  Covariance matrix singular at location [723760,3.7614e+006,0]: skipping...
48: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [725041,3.76141e+006,0]: skipping...
49: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [725041,3.76141e+006,0]: skipping...
50: In predict.gstat(g, newdata = newdata, block = block,  ... :
  Covariance matrix singular at location [725008,3.76144e+006,0]: skipping...

Jean-Romain, 2017-01-19:
Ok I'll make tests on large datasets to understand where does it come from. Maybe it comes from too many data, maybe from negative values, maybe from duplicated points maybe other causes. Could you copy/paste the output of `summary(testlas)`. And could tell if you have duplicated ground values:

```r
duplicated(lasfilterground(lidar)@data, by =  c("X", "Y")) %>%  sum
```
I'm going to investigate on such question.

japolo, 2017-01-19:
> summary(testlas)
class        : LAS
memory       : 255.3 Mb
extent       : 722233.3 , 725202.6 , 3761317 , 3764837 (xmin, xmax, 
ymin, ymax)
area         : 9972705 m^2
points       : 4780516 points
pulses       : 4049474 pulses
point density: 0.48 points/m^2
pulse density: 0.41 pulses/m^2
fields       : 10
field names  : X Y Z gpstime Intensity ReturnNumber NumberOfReturns 
Classification ScanAngle pulseID

File signature:           LASF
File source ID:           0
Global encoding:          1
Project ID - GUID:        0
Version:                  01.02
System identifier:        LAStools (c) by rapidlasso GmbH
Generating software:      las2las (version 160110)
File creation d/y:        131/2010
header size:              227
Offset to point data:     152941
Num. var. length record:  242
Point data format:        01
Point data record length: 28
Num. of point records:    4780516
Num. of points by return: 4048882 687549 43775 310 0
Scale factor X Y Z:       0.01 0.01 0.01
Offset X Y Z:             7e+05 3700000 0
min X Y Z:                722233.3 3761317 187.05
max X Y Z:                725202.6 3764837 747.88
Variable length records:  void

 > duplicated(lasfilterground(testlas)@data, by =  c("X", "Y")) %>%  sum
[1] 4705

I am a small cog in a large research group and we got our data directly 
from U.S.D.A. N.R.C.S. You can access the data at the following link:
https://coast.noaa.gov/inventory/?redirect=301ocm

We used the "Topographic Lidar" data. I don't know how the site 
distributes the LiDAR to other users. I don't use dropbox or anything, 
otherwise, I'd give you the file I've been practicing on. If you have a 
way to grab this from me, I'll be happy to give you a couple of these 
files.

cheers,
John

Jean-Romain, 2017-01-19:
I just processed 4443 k points (36 k ground points) in 11 second with kriging on a regular computer. So it works well. The problems comes from your data. I'm pretty sure the problem comes from duplicated ground points.

> duplicated(lasfilterground(testlas)@ data, by = c("X", "Y")) %>% sum 
> [1] 4705

I'll try on your data

japolo, 2017-01-19:
Okay, This data has had surprises for me before. 

The link works for me. I'm using Firefox. Firefox gives me a warning about outdated Flash in my browser. Maybe your browser doesn't like that the page uses Flash? 

If you search for "United States Interagency Elevation Inventory", you should find it.

Jean-Romain, 2017-01-19:
I disabled my addblockers and other blockers. Works fine. But that does not help me a lot. I have a map of US but I don't know how to download your file and I don't know which file to download. Anyway it does not matter. If the problem is a problem of duplicated points I would be able to reproduce it with dummy data.

Jean-Romain, 2017-01-19:
I reproduced the bug adding a single dummy duplicated point with same X Y Z coordinates. Its easy to fix by checking if the dataset contains duplicated points and by removing them on the fly before to make the interpolation.

But why duplicated points? That *is* the question! 3 possibilities
- The random location of points in overlaps leads to identical coordinates. That sounds very improbable. Especially it occurred 4700 times.
- The file is corrupted
- The number of decimal of the coordinate lead to identical coordinates.

Tell me if duplicated points have the same Z coordinates

```r
duplicated(lasfilterground(testlas)@data, by = c("X", "Y", "Z")) %>% sum
```

Show me the digits of your coordinates

```r
options(digits = 20)
head(testlas$X)
```

japolo, 2017-01-19:
The data came in bulk, so to say, so I don't know how to point to 
individual tiles. This link is for the metadata:
https://coast.noaa.gov/data/Documents/Metadata/USIEI/OK_DamRehabAssessmentAML1_2012_Metadata.xml

This could probably help find the geographic extent on the starting 
page, but it won't get you the files, just narrow the search.

Anyway, I don't know if I am helping you any more. It seems like you 
have it working, it's a problem on my side. Thanks for your help. 
Hopefully, in the future I'll get to use this with appropriate data.

cheers,
John

japolo, 2017-01-19:
On 1/19/2017 10:50 AM, Jean-Romain Roussel wrote:
>
>   * The random location of points in overlaps leads to identical
>     coordinates. That sounds very improbable. Especially it occurred
>     4700 times.
>
Do you know about multiple returns / pulse? That's what this sounds like 
to me.

>   * The file is corrupted
>
That's possible, but I don't think it's likely.

>   * The number of decimal of the coordinate lead to identical coordinates.
>
> Tell me if duplicated points have the same Z coordinates
>
> duplicated(lasfilterground(testlas)@data,by  =  c("X","Y","Z")) %>%sum
 >  duplicated(lasfilterground(testlas)@data, by = c("X", "Y", "Z")) %>% sum
[1] 717

> options(digits  =  20)
> head(testlas$X)
 > head(testlas$X)
[1] 724856.66000000003 724860.26000000001 724858.29000000004 
724884.34999999998
[5] 724881.06999999995 724880.89000000001

Jean-Romain, 2017-01-19:
It's also a problem to me. My code must handle existing problems. Your problem is an existing problem. So I must, at least, raise a readable error. But the best would be to treat the problem properly. That's why I'm wondering why you have duplicated points

> duplicated(lasfilterground(testlas)@data, by = c("X", "Y", "Z")) %>% sum 
> [1] 717

You have 717 ground points with the same X Y Z coordinates. Definitively weird.

japolo, 2017-01-19:
On 1/19/2017 11:00 AM, Jean-Romain Roussel wrote:
> You have 717 ground points with the same X Y Z coordinates. 
> Definitively weird.
I guess that is. The other file I was using for practice has 677 
duplicates. Should it necessarily be a problem? I've tried other 
software, like QTModeler and LP360, with these tiles and they didn't 
seem to get hung up on the issue.

Jean-Romain, 2017-01-19:
It depends on the algorithm used. You can use the inverse distance weighting interpolation. The algorithm will work without any trouble. But does it makes sense? Let suppose you have 5 duplicated points with the coordinates A(0.8,0.9). You want to get the interpolation at  B(1,1). The 5-nearest neighbors are these 5 points A instead of the 5 points around B . So the algorithm can compute the solution. No trouble. But does it make sense? No.

So yes other softwares can solve the problem. But what does it means? It depend on the algorithm. With a Delaunay triangulation it does no change anything. With a knn approach that's lead to unexpected results

Worse, if all the duplicated points have the same coordinates we can retain only one of them (what `lidR`does now). But if you have 2 ground points A(1,2,10) and B(1,2,11) which one is the wrong one?

Jean-Romain, 2017-01-19:
Erratum: it does not work with a Delaunay triangulation. Actually you must choose only one of the duplicated point or you will have some 0 area triangles.

Jean-Romain, 2017-01-19:
New behavior

```r
> dtm = grid_terrain(las, method = "xxxx", k = 10)
Warning messages:
1: 8 duplicated ground points with the same X Y Z coordinates were removed. 
2: 3 duplicated ground points with the same X Y coordinates but different Z coordinates. min Z were retained 
```
Sounds good?

japolo, 2017-01-19:
On 1/19/2017 3:19 PM, Jean-Romain Roussel wrote:
>
> New behavior
>
> >  dtm  =  grid_terrain(las,method  =  "xxxx",k  =  10)
> Warning  messages:
> 1:  8  duplicated  ground  points  with  the  same  X  Y  Z  coordinates  were  removed.
> 2:  3  duplicated  ground  points  with  the  same  X  Y  coordinates  but  different  Z  coordinates.min  Z  were  retained  
>
> Sounds good?
>
Yes! Tres bien!