r-lidar/lidR #40: writeLAS issue

tochaf, 2017-02-01:
Hi, and tank you for this useful function.

The writeLAS is not working properly because it mixes the attributes. For example, if you do 

lidar<-readLAS(files="a.laz")   
writeLAS(lidar,"a_write.laz")

you will find the Intensity in the gpstime slot etc. I believe it downs not take the attributes in the good order.

Jean-Romain, 2017-02-01:
Which version are you using?  I already solved that issue once.

tochaf, 2017-02-02:
Version "1.1.0.9000" freshly installed.

Jean-Romain, 2017-02-02:
Actually I didn't try it yet, I was busy but I will run it tonight. I only wanted to make sure that you do not use an old version because I already solved such bug some month ago. I you are using the lastest version there is a real different problem.

Jean-Romain, 2017-02-03:
As for the other issue you opened I cannot reproduce the bug.

```r
library(lidR)
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")

las = readLAS(LASfile)

writeLAS(las, "~/Bureau/test.las")

las2 = readLAS("~/Bureau/test.las")

las$data

              X       Y     Z  gpstime Intensity ReturnNumber NumberOfReturns Classification ScanAngle pulseID
    1: 684992.2 5018007 17.30 483825.9        41            1               1              1         5       1
    2: 684992.6 5018006 17.03 483825.9        54            1               1              1         5       2
    3: 684993.0 5018005 16.14 483825.9        51            1               2              1         5       3
    4: 684993.1 5018006 11.10 483825.9         3            2               2              1         5       3
    5: 684993.2 5018001 17.58 483825.9        11            1               3              1         5       4
   ---                                                                                                        
81586: 684945.1 5018004  2.45 484376.7         3            4               4              1        16   56975
81587: 684944.3 5018007 13.20 484376.7         3            2               2              1        16   56976
81588: 684946.4 5018005  2.40 484376.8         4            3               3              1        16   56977
81589: 684947.4 5018006  0.00 484376.8         6            3               3              2        16   56978
81590: 684947.2 5018007  0.86 484376.8         5            2               2              1        16   56979

las2$data

             X       Y     Z  gpstime Intensity ReturnNumber NumberOfReturns Classification ScanAngle pulseID
    1: 684992.2 5018007 17.30 483825.9        41            1               1              1         5       1
    2: 684992.6 5018006 17.03 483825.9        54            1               1              1         5       2
    3: 684993.0 5018005 16.14 483825.9        51            1               2              1         5       3
    4: 684993.1 5018006 11.10 483825.9         3            2               2              1         5       3
    5: 684993.2 5018001 17.58 483825.9        11            1               3              1         5       4
   ---                                                                                                        
81586: 684945.1 5018004  2.45 484376.7         3            4               4              1        16   56975
81587: 684944.3 5018007 13.20 484376.7         3            2               2              1        16   56976
81588: 684946.4 5018005  2.40 484376.8         4            3               3              1        16   56977
81589: 684947.4 5018006  0.00 484376.8         6            3               3              2        16   56978
81590: 684947.2 5018007  0.86 484376.8         5            2               2              1        16   56979

closed 2017-02-03 by Jean-Romain