r-lidar/lidR #41: lasdecimate issue

tochaf, 2017-02-01:
Hi

Many functions do not work properly (e.g. writeLAS, lasdecimate) because it switches the attributes. For example, within the lasdecimate, the "intensity" is removed and its values assigned to the "gpstime" attribute.

I'm wondering whether this issues arent' related to the fact that we are using  lastools without license.

Jean-Romain, 2017-02-01:
What is the problem? I mean if `writeLAS` function does not work well then if you read back the corrupted file it's normal than the other functions fail. So the problem is `writeLAS` not `lasdecimate`

Am I right ?

tochaf, 2017-02-02:
Did not read the corrupted file (i.e. the one saved with writeLAS). I read the original .laz and run the "lasdecimate" without saving it (i.e did not use writeLAS) and the objects that shows up in the R workspace produced by "lasdecimate" has the attributes in the wrong slot .... for example, "number of returns" appear twice in "intensity" and "classification" or something similar

Jean-Romain, 2017-02-02:
You ran

```r
las = readLAS(...)
thinned = lasdecimate(las)
```

And `thinned` was corrupted :open_mouth:

Ok, that's weird. I will try.

tochaf, 2017-02-02:
yes, have run that .. let me know the results in your machine, thanks

Jean-Romain, 2017-02-02:
I ran that code without issue. Could you copy/paste exactly what you did.

```r
library(lidR)
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
lidar  = readLAS(LASfile)
thinned = lidar %>% lasdecimate(0.5, res = 5)

lidar$data

            X       Y     Z  gpstime Intensity ReturnNumber NumberOfReturns Classification ScanAngle pulseID
    1: 684992.2 5018007 17.30 483825.9        41            1               1              1         5       1
    2: 684992.6 5018006 17.03 483825.9        54            1               1              1         5       2
    3: 684993.0 5018005 16.14 483825.9        51            1               2              1         5       3
    4: 684993.1 5018006 11.10 483825.9         3            2               2              1         5       3
    5: 684993.2 5018001 17.58 483825.9        11            1               3              1         5       4
   ---                                                                                                        
81586: 684945.1 5018004  2.45 484376.7         3            4               4              1        16   56975
81587: 684944.3 5018007 13.20 484376.7         3            2               2              1        16   56976
81588: 684946.4 5018005  2.40 484376.8         4            3               3              1        16   56977
81589: 684947.4 5018006  0.00 484376.8         6            3               3              2        16   56978
81590: 684947.2 5018007  0.86 484376.8         5            2               2              1        16   56979

thinned$data

              X       Y     Z  gpstime Intensity ReturnNumber NumberOfReturns Classification ScanAngle pulseID
    1: 684992.2 5018007 17.30 483825.9        41            1               1              1         5       1
    2: 684992.6 5018006 17.03 483825.9        54            1               1              1         5       2
    3: 684993.0 5018005 16.14 483825.9        51            1               2              1         5       3
    4: 684993.1 5018006 11.10 483825.9         3            2               2              1         5       3
    5: 684992.4 5018002 18.85 483825.9        31            1               2              1         5       6
   ---                                                                                                        
33716: 684944.6 5018007 16.34 484376.7        25            2               3              1        16   56974
33717: 684945.3 5018004  2.49 484376.7         5            3               3              1        16   56974
33718: 684945.1 5018004  2.45 484376.7         3            4               4              1        16   56975
33719: 684946.4 5018005  2.40 484376.8         4            3               3              1        16   56977
33720: 684947.2 5018007  0.86 484376.8         5            2               2              1        16   56979

Jean-Romain, 2017-02-02:
> I'm wondering whether this issues arent' related to the fact that we are using lastools without license.

I used LASlib which is the open source part of LAStools to read and write las and laz files. All the other functions were written from scratch in R or C++. I do not use LAStools.

tochaf, 2017-02-02:
you are right. Just run the lasdecimate with a couple of .las and .laz files (including "MegaPlot.laz") and the function seems to be working properly. My fault I guess :)

Jean-Romain, 2017-02-02:
It's never the user' fault. If you encountered a bug it means that something is wrong. The user can create but only because the developer didn't provide strong enough code.

If you succeed in reproducing the bug tell me.

closed 2017-02-03 by Jean-Romain