r-lidar/lidR #43: lidR_fast_countbelow

pdbentley, 2017-02-06:
Hi Jean-Romain, 

I've been working with your package again - thank you for efforts. I have been working on my laptop and things are working fine (windows 10), and just did a fresh installation on a desktop computer (also windows 10), however i now get the following error:

LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
lidar = readLAS(LASfile)

Error in .Call("lidR_fast_countbelow", PACKAGE = "lidR", x, t) : 
  "lidR_fast_countbelow" not available for .Call() for package "lidR"

The version of lidR on my laptop was installed about a month ago; the desktop just then.  

Thoughts?

Jean-Romain, 2017-02-06:
Hi

I hope the problem is only a problem of old development version because I just released the new version on CRAN.

Tell me which version you are using.

```
packageDescription("lidR")$Version
```

And update your version (from github because the CRAN didn't built the new Windows binaries yet).

```r
devtools::install_github("Jean-Romain/rlas", dependencies=TRUE)
devtools::install_github("Jean-Romain/lidR", dependencies=TRUE)
```

pdbentley, 2017-02-07:
Hi, 

Both my laptop and desktop are using version 1.1.0, and both were installed from github. 

But, its working now. Yesterday i tried to uninstall the packages using remove.packages() and then reinstall them from github but that didn't work (i still had an earlier version of lidR on my desktop). Today i physically deleted the folders for 'rlas', 'lidR', and 'Rcpp' (the last one as a punt), and then reinstalled. And now it works. So it was an installation issue / conflict. 

Thanks for you assistance.

closed 2017-02-07 by pdbentley