r-lidar/lidR #45: Bug in lastrees on CRAN for windows oldrel

Jean-Romain, 2017-02-08:
See [lidR results](https://cran.r-project.org/web/checks/check_results_lidR.html). Not able to reproduce the bug.

```
running examples for arch 'x64' ... ERROR
Running examples in 'lidR-Ex.R' failed
The error most likely occurred in:

> ### Name: lastrees
> ### Title: Individual tree segmentation
> ### Aliases: lastrees
>
> ### ** Examples
>
> LASfile <- system.file("extdata", "Tree.laz", package="lidR")
> las = readLAS(LASfile, XYZonly = TRUE, filter = "-drop_z_below 0")

[...]

> # segmentation (default parameters but th = 4 + extra output)
> extra = lastrees(las, "dalponte2016", chm, th = 4, extra = TRUE)
Error in as.vector(data) :
  no method for coercing this S4 class to a vector
Calls: lastrees -> dalponte2012 -> matrix -> as.vector
Execution halted
```

shafrina, 2017-02-14:
Hi,

Could you please add function how to write individual tree boundary to shapefile after the lastrees segmentation?

Jean-Romain, 2017-02-14:
Hi, could you please open a specific issue with more details about what you're expected. There are many way to achieve that.

shafrina, 2017-02-14:
Hi,

I need to convert the crown segmented to shapefiles format (ESRI shapefiles). I know this could be achieved in a few ways, like using writeogr function from rgdal package or using maptools, but i don't know how to start after get the crown (after run the lastrees). My R skills is kinda basic. The result is pretty convincing, but i need to get it analyse in arcgis to compare it with my field crown.

Jean-Romain, 2017-02-14:
(using the example given in the doc)

At the end, if you chose a method that enable for an output of the crown in raster format (i.e. `dalponte` or `wartershed` using `extra  = T`). Then the `raster` package have all the tools you need to process and analyse you rasters. You should look at `rasterToPolygon`

```r
extra = lastrees(las, "dalponte2016", chm, th = 4, extra = TRUE)

library(raster)
crown.shp <- rasterToPolygons(extra$Crown, dissolve = TRUE)
plot(crown.shp)
```

You can also convert that into convex hulls with `rgeos`

```r
crown.hull <- rgeos::gConvexHull(crown.shp, byid = T)
```

Then, `writreOGR` enable for writing a shapefile.

I'm not a specialist. You should look for better information from people who know more than me about these subjects. I can't teach you `raster` `sp` `rgeos` or `maptools`  but I know that all you need already exists.

shafrina, 2017-02-15:
Hi,

Yes it works to transfer the raster to spatial polygon dataframe. Thanks so much.
i just simply run it as
crown.shp <- rasterToPolygons(extra, dissolve = TRUE)
and to save it as shapefiles:
library(maptools)
writePolyShape(crown.shp, "extdata.shp")

It doesn't delineate correctly when i compare it with field data. But anyhow, if adding morphological gradient it will improve the delineation especially my forest is tropical forest. I'm trying to find the coding in R to add the morphological formula to aid with the watershed method.

commit dd3a0513 (2017-02-13): Expected to fix #45
commit beb4817d (2017-02-14): Expected to fix #45 for watershed
closed 2017-02-19 by Jean-Romain