r-lidar/lidR #51: Forced lowest point in pixel in grid_terrain

jmmonnet, 2017-02-24:
I think it would useful for the user to be able to choose whether the lowest point in the pixel should be chosen in the grid_terrain function or not. Indeed, in slope areas, depending on the sampling pattern of lidar points, this feature would lead to systematic bias (e.g. if there are several lidar points per cell, the downhill point will be systematically selected, leading to a lower elevation in the case of a regular slope) or artefacts (e.g if there is one or less point in each cell, this would create some additional roughness in the case of a regulare slope) compared to a simple delaunay triangulation.

Jean-Romain, 2017-02-24:
Hi, 

That makes sense. I'm going to relax this constrain adding an optional parameter. Actually I did like that to avoid too many negative values.

You can also used a strict interpolation of each point without raster (since 1.2.0).

Jean-Romain, 2017-02-24:
Done. There is now an option `keep_lowest`. Thanks for contributing.

closed 2017-02-25 by Jean-Romain