r-lidar/lidR #53: Unable to create the Normalize dataset

karnayogendra, 2017-03-05:
Hi ,

I was trying to create the normalize dataset of R, but every time I found the error stating that "Warning message: Dataset may be invalid: 74056 points below 0 found."  I don't know how to remove or filter the points below 0 so that I can create the normalize .las dataset.

Regards,
Yogendra

Jean-Romain, 2017-03-05:
Hi, 

This is a warning not an error. Actually it is nearly impossible to normalize a point cloud without warning especially if you used a raster as digital terrain (v1.2 enable for normalization without raster).

Using a raster as digital terrain the elevations of points classified as ground can be greater or lower than 0 so there are many points below 0. Using an exact interpolation you ensure that all ground points are exactly at 0 but you can still have points below 0, it depends on how the ground points were classified.

[image]

To remove negative values you can use a filter

```r
las = lasfilter(las, Z >= 0)
```

karnayogendra, 2017-03-05:
Thanks for your prompt response.

Regards,
Yogendra

*...............................*
*Yogendra K. Karna*
*PhD Candidate*
*School of Ecosystem and Forest Sciences*
*The University of Melbourne*

4 Water St., Creswick, VIC 3363, Australia

T: +61 3 5321 4257

M: 0410 207 191

*E: ykarna@student.unimelb.edu.au <ltb@unimelb.edu.au>*

closed 2017-03-05 by Jean-Romain