r-lidar/lidR #54: writelas error

muzo583, 2017-03-14:
When I ve created LAS file from data frame, after tried to export LAS object, I see error below. What is the error? thanks

`writeLAS(test,"test.las")'

index out of bounds

Jean-Romain, 2017-03-14:
Hi

How did you create your `LAS` object from a `data.frame`? Without any minimal reproducible example I can't answer your question. But I know it comes from LASlib and I'm sure there is a good reason why it failed to write your data but I need more information abut the data.

muzo583, 2017-03-14:
It includes X,Y,Z,Gpstime and Classification. when I create LAS object, LAS function does not give any error,

```
test <- LAS(data.frame)
 writeLAS(tes, "test.las")

```
I don't know what to do with the data any some specific operation or extra information?

I tried also  this 

```
groundT2 <- readLAS("L:/g_duz.las")
groundT2 <- as.SpatialPointsDataFrame(groundT2)
groundT2 <- data.frame(groundT2)
groundT2 <- LAS(groundT2)
writeLAS(groundT2,"groundT.las")
```
and same error..

test data is here [test data](https://drive.google.com/file/d/0B9CZcJOo5aFoNURNcWx0YUlVRm8/view?usp=sharing
)

Thank you Romain.

Jean-Romain, 2017-03-14:
Ok, I'm going to test that today.

Jean-Romain, 2017-03-14:
I found the bug, it's and old bug already known but I completely forgot it. It's not a bug from `lidR` but from `rlas`.

What happened? You created a `LAS` object from a `data.frame`

```
groundT2 <- LAS(data.frame)
```

But this object `groundT2` contains only a partial header. This header contains the number of point recorded, the number of first returns, the bounding-box of the data and so on because it can easily be computed from the data but it cannot guess neither for the offset nor the scale factor nor the point type. These information are missing. If you do not understand what I'm talking about please refer to the [LAS specifications](http://www.asprs.org/wp-content/uploads/2010/12/LAS_1_4_r13.pdf) )

I'll solve this bug by attributing default values if these fields are empty. But in my opinion it's not a good solution. What you can do is to copy the header of the another las file (the original one preferably).

```
groundT2 <- readLAS("L:/g_duz.las")
oheader <- groundT2@header

groundT2 <- as.SpatialPointsDataFrame(groundT2)
groundT2 <- data.frame(groundT2)
groundT2 <- LAS(groundT2, oheader)

writeLAS(groundT2, "groundT.las")
```

Jean-Romain, 2017-03-14:
There is another bug. All the field are not written. I'm investigating.

Jean-Romain, 2017-03-14:
Commit  47373d4 enable for computing the whole header automatically. The `LAS` constructor automatically detect the point type and the attribute default value for offsets and scales  It means that the file will be properly written without bug. However a warning will be raised for each important value attributed by default

```
groundT2 <- as.SpatialPointsDataFrame(groundT2)
groundT2 <- data.frame(groundT2)
groundT2 <- LAS(groundT2)
Warning messages:
1: Dataset is invalid: 6897 points with a return number of 0 found. 
2: 'X offset' was missing in the header. The value 455685.05 was automatically attributed 
3: 'Y offset' was missing in the header. The value 4088335.68 was automatically attributed 
4: 'Z offset' was missing in the header. The value 1319.13 was automatically attributed 
5: 'X scale factor' was missing in the header. The value 0.01 was automatically attributed 
6: 'Y scale factor' was missing in the header. The value 0.01 was automatically attributed 
7: 'Z scale factor' was missing in the header. The value 0.01 was automatically attributed 

writeLAS(groundT2,"~/Bureau/groundT.las")
234725 bytes written
```

muzo583, 2017-03-15:
Dear Romain,

Thank you for your effort and great package. Now it is fixed !

Cheers.

closed 2017-03-15 by Jean-Romain