r-lidar/lidR #55: [bug] lastree method "li2012" provide EMPTY output

spono, 2017-03-23:
lasground-> grid_terrain-> lasnormalize-> lasfilterfirst-> lastree-> empty output.
Tested the tool from 1.2.0.900 version on three different machines with Windows 10 64bit .

Jean-Romain, 2017-03-25:
Hi,

Than you for reporting this bug. Did you test with the data given as example? It should work. If it doesn't I'll investigate. If the method works with the data given as example but not with yours I will need a sample of your data which enable to reproduce the bug. Otherwise I'll not be able to find the bug.

Thank you

Jean-Romain, 2017-03-26:
Mmm forgot my previous message, actually I think this is not a bug, its a feature. The default behavior of `lastrees` is to return nothing and update your original dataset by reference. Please refer to the doc. For the two other algorithms when `extra = true` you can get an optional output but this is not the default behavior.

closed 2017-03-29 by Jean-Romain