r-lidar/lidR #56: [UI-enhancement] progress bar

spono, 2017-03-23:
In order to keep track if the pc is properly working, it would be nice to add the progress bar (such as for lastrees) also to the other tools.

Jean-Romain, 2017-03-25:
Hi,

I'm sorry it's not possible because many tools rely on functions provided by R or by other packages. To change the behavior of the functions I have to write my own functions. In lastrees, each method relies on my own code but (for example) I wrote neither the delaunay triangulation nor the data.table aggregation. I don't have any control while they are running.

Jean-Romain

Jean-Romain, 2017-05-02:
I can't add progress bar everywhere as I explained previouly but `v1.2.1` will enable for a verbose mode.

```r
lidr_options(verbose = TRUE)
```

Jean-Romain, 2017-05-07:
In addition there is now an option

```r
lidr_options(progress = TRUE)
```

To display progress bar when available. Several new progressbar are available. In addition several algorithm are 50 to 100 times fasters such as Delaunay rasterization.

spono, 2017-05-08:
(still missing in the Help, right?)
btw; wonderful, thank you very much!

Jean-Romain, 2017-05-08:
`?lidr_options` should work (if you installed the very last version)

spono, 2017-05-08:
sorry, I was referring just to the 'progress' switch.
BTW, testing the latest, it gives me "In LIDROPTIONS(...) : Ignoring options not defined in manager: progress"

Jean-Romain, 2017-05-08:
It is currently only available in the `devel` branch. I didn't merge the `devel` branch into `master` yet. 
FYI v1.2.1 will probably be released on CRAN this week on the next one.

closed 2017-05-13 by Jean-Romain