r-lidar/lidR #58: Error in lasground

karnayogendra, 2017-04-15:
Hi,  I found following error while using lasground for Tree-segmentation-from-A-to-Z exercise from the following link: https://github.com/Jean-Romain/lidR/wiki/Tree-segmentation-from-A-to-Z.

Error in .Call("lidR_MorphologicalOpening", PACKAGE = "lidR", X, Y, Z,  : 
  "lidR_MorphologicalOpening" not available for .Call() for package "lidR"
Please advise me how to solve it.

Thanks and Regards,
Yogendra

Jean-Romain, 2017-04-15:
Hi,

This not the first time somebody report a strange problem like that. It cannot be a problem with the code. I mean, all the examples of code from the doc are tested by the CRAN (and I) and also other tests are ran. If a function was missing (or was not callable) the package would be rejected. Therefore I guess it's a problem with the user installation. So try to uninstall and reinstall the package. If the problem persist we will investigate further.

JR

Jean-Romain, 2017-04-15:
See also #43

closed 2017-04-18 by Jean-Romain