r-lidar/lidR #59: methods instaead of functions

floriandeboissieu, 2017-05-02:
Hi Jean-Romain,
the use of stopifnotlas is preventing from any class inheritance. Is there any reason you do not define them as methods for LAS object?
Cheers,
Florian

Jean-Romain, 2017-05-02:
Yes there are severals reason and no one is good. 

Methods are boring! It double the length of the table of content in the documentation, it's boring to write (more code) and it's useless if your functions are designed for a single type of object. At the beginning I was using S4 methods. But it was boring. From this old code it remains the LAS object which is S4.

So my explanations are not good at all I agree. If you want to discuss about the design and give me advices I'm ok with that. There is a real problem of design especially for inheritance...

floriandeboissieu, 2017-05-02:
I am no expert in OO programing especially for R, so wouldn't have any 
advice. I noticed in other well known packages, like package 'raster', 
that most were methods. And as I wanted to enhance the class LAS with 
other informations that are not in the standard. Thus creating an 
inherited class would have been a good way not to have to rewrite the 
methods, although there are other ways around.

But I did not noticed the table of content problem, which I have to 
admit is quite anoying.

Jean-Romain, 2017-05-02:
In `raster`or `sp`, S4 classes make a lot of sense. In `lidR` not really. As I said S4 classes for LAS objects comes for older code when I discovered  R OO style. But S4 classes are not really C++ OO style programming and it was more boring than useful. R6 classes are more like C++ OO. If I had to begin again I probably would use R6 classes. 

I think I will have a bit of pain to change that without any compatibility trouble. That is why I never changed that and LAS remains an S4 class. But I can work on it if you want. Tell me what you want do to, I'll try to think about it.

floriandeboissieu, 2017-05-02:
It was more an open question than a question on a specific purpose. I 
tried to extend LAS class, and thus I thought it would have been easier 
with inherited methods. But I understand your point of view and actually 
I found a way around that is sufficient for the moment. So no worries 
about the R6 classes.
Cheers

Jean-Romain, 2017-05-04:
Well I thought about it. I think you can modify the function `stopifnotlas` by using `isClass` instead of `class(x)[1]`.  It should tests for inheritance. Try it and tell me (or make a PR)

Cheers

Jean-Romain, 2017-05-06:
I think this commit will solve the problem

floriandeboissieu, 2017-05-06:
Ok, thanks, sorry I didn't have the time to help this week...

closed 2017-05-11 by Jean-Romain