r-lidar/lidR #61: writeLAS and `Header Size`

karnayogendra, 2017-05-08:
Hi, 

I have a tile with a 'header size` of 235. However when i save this file using writeLAS it is changing the 'Header size` to 227. This is creating errors when i re-import the tile: 

WARNING: for LAS 1. header_size should at least be 235 but it is only 227
WARNING: for LAS 1. header_size should at least be 235 but it is only 227
WARNING: for LAS 1. header_size should at least be 235 but it is only 227
WARNING: for LAS 1. header_size should at least be 235 but it is only 227
Warning messages:
1: Invalid file: header states the file contains 11221946 1st returns but 16096836 were found. Header has been updated. 
2: Invalid file: header states the file contains 8310139 2nd returns but 9559154 were found. Header has been updated. 
3: Invalid file: header states the file contains 4354005 3rd returns but 0 were found. Header has been updated. 
4: Invalid file: header states the file contains 1439017 4th returns but 0 were found. Header has been updated. 
5: Invalid file: header states the file contains 288440 5th returns but 0 were found. Header has been updated. 

I have tried to re-import the file using LASTools, so I suspect the header size is being changed during writeLAS. The result is a corrupted file when saving.

The tile causing the issue is new (Version 1.3).  We spoke with Martin Isenburg and he recommended the writeLAS needs to be set at Version 1.2 to avoid this error. We are wondering what the current setup is for writeLAS in the lidR package. 

Thanks.

Jean-Romain, 2017-05-08:
Hi,

Thank you for reporting the bug. Please attach an example file and an example code. I must make some test myself to find the problem.

Regards

karnayogendra, 2017-06-02:
Hi Jean,

My apology for the delayed reply. I was on leave just after confirmation of
candidature of PhD so could not respond to your email. I need permission to
send the example data sets so once I obtain it will send it to you however
the file size is big though.

Thanks and Regards,
Yogendra

*...............................*
*Yogendra K. Karna*
*PhD Candidate*
*School of Ecosystem and Forest Sciences*
*The University of Melbourne*

4 Water St., Creswick, VIC 3363, Australia

T: +61 3 5321 4257

M: 0410 207 191

*E: ykarna@student.unimelb.edu.au <ltb@unimelb.edu.au>*

Jean-Romain, 2017-06-02:
No problem. A little subset of it should be enough. What I need is only a toy example with the proper metadata to reproduce the problem not the whole dataset.

karnayogendra, 2017-06-06:
Hi Jean,

Here is the subset of one tile that I have clipped for 100*100 m. After that I exported the clipped tile as another las file, which is written as test_write_las.las but while I try to read it shows the message as of first email. 
The code I have used are:
`Infile <- "F:/PilotTest_Confirmation/data/LasFiles/e387n5847_centralhighlands_2016jan09_mpts-c2_v20cm_ahd-mga55.las"`
`Infile`
`ReadLas <- readLAS(Infile)`
`clip1 <- lasclipRectangle(ReadLas, 387593, 5847324, 387693, 5847424, inside = T)`
`clip1`
`writeLAS(clip1, file = "F:/PilotTest_Confirmation/data/LasFiles/test_write_las.las")`
`Rtest <- readLAS("F:/PilotTest_Confirmation/data/LasFiles/test_write_las.las")`

The sample subset of clipped file is attached herewith.

[testfile.zip](https://github.com/Jean-Romain/lidR/files/1054395/testfile.zip)

Jean-Romain, 2017-06-06:
Well, I'm really grateful for your help but you provided me the output files (the corrupted one). What I need is the original file to reproduce the bug. Indeed I perfectly understand the problem but I don't have a las file in version 1.3 to reproduce the problem. And I'm not able to understand it only reading the code. Indeed the `header_size` is not expected to be modified. If you don't want to send me an original fileI understanf but please copy the output of `lasinfo` (from LAStools or liblas) or simply `print(ReadLas@header)`)

Regards

karnayogendra, 2017-06-07:
Hi Jean,

Please find the link for data download.
[](https://www.dropbox.com/sh/mwp56ewohkrrdd5/AABHDD-Xq45imilaEmCWd-fRa?dl=0)

If it will not work, please let me know.

Thanks and Regards,
Yogendra

Jean-Romain, 2017-06-07:
It was easy to find the bug with the good data. It was not a bug with `lidR` but with `rlas`. Please update [rlas](https://github.com/Jean-Romain/rlas). 

Notice that the way I write las file is not perfect. There are some informations lost from the original file.
- The variable length records which usually contains information about the projection. I didn't figure out how to write it yet.
- The extra data. To make it simple the data provider can write anything he want between the header and the data (points). To do so the header should explicitly contains the positioning of the point in the file. I don't know how to read these extra info and how to write them. So the output file skip these informations.

But everything else sound good.

closed 2017-06-10 by Jean-Romain