r-lidar/lidR #63: qhull error using pitfree algorithm

spono, 2017-06-21:
using the pitfree alg on a normalised las, it returns the following error:

QH6114 qhull precision error: initial simplex is not convex. Distance=-4.1e-09
Error in geometry::delaunayn(X[, 1:2]) : 
  Received error code 2 from qhull.

I don't know if it may be related to too few data per single height slice.
how is it possible to overcome this issue?

Jean-Romain, 2017-06-23:
Hi,

Thank you for reporting the bug and sorry for the delay, I was in vacation. This bug is related to the `geometry` package. Can you provide me a reproducible example (code + data)? According to the qhull documentation it is likely to be a bug with the accuracy of you data and it can be solved by passing the good options to the delaunay triangulation function. I would like to reproduce the bug to get a more specific idea of the bug. Then I will asked @davidcsterratt what is the best way to solve it.

Jean-Romain, 2017-06-24:
I'm not able to reproduce the bug since I got another bug related to qhull too.

> QH6227 qhull precision error: Only 4 facets remain.  Can not merge another
> pair.  The input is too degenerate or the convexity constraints are  too strong.
> Error in geometry::delaunayn(X) : Received error code 1 from qhull.

I contacted @davidcsterratt and the discussion moved [here](https://github.com/davidcsterratt/geometry/issues/12).

spono, 2017-06-24:
Hi,
sorry I realised only now that with the whole cloud it returns the “error 1” you reported while, using the normalised one, returns the “error 2” I described above.

Jean-Romain, 2017-06-26:
Hi,

I added an option in the Delaunay triangulation to rescale the point cloud and fit it between [-0.5 ; 0.5]. This enable to solve your precision issue. I think it might be safe in general but I must ensure it does not create new bugs. Currently it works fine on my tests.

You can download the version 1.3.0 in the development branch.
```r
devtools::install_github("Jean-Romain/lidR", dependencies=TRUE, ref = "devel")
```
But please do not use the new features relative to catalogs. This is still in heavy development. I cannot guarantee it will work fine and I cannot guarantee it won't change before the release of v1.3.0

spono, 2017-06-29:
Wonderful, it works perfectly: thank you very much!

Some possbile enhancements I was thinking to in the last month:
-  a tool to define the slope (from the identified tree top to the margins of the crown) of the segmented crown;
- a tool that operates on the segmented cloud to provide basic description parameters on a plot basis (i.e. given the centre plot and the radius, it returns the number of trees/ha, avg height, dominant height, or some structural indeces)

In relation to the indeces (e.g. diversity indices), here you may find some other ideas or R/C++ scripts that can be integrated as functions.

http://www.pommerening.org/wiki/index.php?title=Arne_Pommerening%27s_Forest_Biometrics_Lab#Selected_R_scripts_and_C.2B.2B_files

All the best,

Niccolò

Jean-Romain, 2017-06-29:
I think `tree_metrics(las, .stdtreemetrics)` make, at least partially, what you're looking for in the second point. Your're first point can be computed with the same `tree_metrics` function using a user define function but currently there is no built-in function since this task is no trivial.

Thank you for the link I will look at it. It sounds interesting.

closed 2017-06-29 by Jean-Romain