r-lidar/lidR #64: Readme argument inconsistency

swinersha, 2017-06-26:
Just been going through the read me and noticed that the argument res is listed as resolution for the function lasdecimate. 

Tom

Jean-Romain, 2017-06-26:
Hi,

Are you talking about [this page](https://github.com/Jean-Romain/lidR/wiki/lasdecimate) on the wiki? If yes, I made the correction. This page was an old one an not up-to-date for a long time. Thank you for reporting the error. 

Also notice that the wiki is... a wiki. It means that everybody can contribute and make some change. Feel free to modify it if you encounter some other errors. Your contribution will be appreciated.

swinersha, 2017-06-26:
Indeed. Just thought I’d point it out. 

I’m still a little bit of a newb on github. Appreciate the hint about the wiki. Will make changes in future. 

Cheers.

Tom

Jean-Romain, 2017-06-26:
Yes it fine and appreciated. You can report it in an issue as well. It's only a suggestion. Fell free to report the error in the way you want. Whatever the way it helps me and your contribution is appreciated.

closed 2017-06-26 by Jean-Romain