r-lidar/lidR #66: Plot decimation

swinersha, 2017-06-26:
When plotting it seems that by default all points are plotted, which seems a little impractical for even moderately sized las files. It might be better to decimate as standard for files over a certain size and include an argument to override this.

Tom

Jean-Romain, 2017-06-26:
Hi,

You're right all points are plotted. And you're right `rgl` has some difficulties to display large point clouds. I already thought about the decimation of the point cloud but I don't think it is the best idea. Actually `OpenGL` should be able to deal with large number of points. But I guess that `rgl` uses some old `OpenGL` features and does not take advantage of new methods provided by `OpenGL`. I cannot be sure of that. It's only a guess based on how other softwares work and on some reading on computer science forum.

So I would like to communicate with the `rgl` team to get more informations on this subject. But at the time being I never worked further on this issue.

Feel free to propose a pull request with an implementation of a decimation before plotting. I'm ok to implement this feature but I'm not going to do it myself. Indeed if I begin to work on this issue I prefer to look at `rgl` instead.

Regard

Jean-Romain, 2017-09-30:
I'm curruently working on a new R package which can handle much much more point comfortably :wink:

bi0m3trics, 2017-09-30:
Just a thought, but being able to "add" plots would be a nice feature. For instance, being able to plot a las and then add a second las and maybe even add the dtm to the plot would be awesome for visulization purposes. 

I realized that this could be done by adding id's and coloring, but sometimes you might want to "build" a plot based on different objects. Also, if it's based off rgl - lighting effects could help (eye-dome lighting filters help with vosulizing depth).

Looking forward to seeing what you produce.
-a

bi0m3trics, 2017-09-30:
But then again, cloudcompare does that easily enough...

Jean-Romain, 2017-09-30:
Is there a way to call cloud compare in R? (I'm not talking about system call).

I'm looking for a full R solution and memory efficient. I'm creating a new package for that. `rgl` is great but can't handle millions of points. Try to display 10 millions points using the current `plot` function, you can't use it. My new package can easily.

Advantage of my package:
- handle easily more than 10 million points while `rgl` cannot deal up to a 1/10 of that.
- have pan zoom rotate while `rgl` only have zoom rotate.
- much memory efficient.

Drawback
- Run on the R session. So if the windows is open, you can't make R stuff. You have to close the windows (so no multiple plots)
- It enables only to display a point cloud. No other feature. 

So an `add` feature is not in currently in my plan. My plane is to reproduce the current function `plot` but with an instantaneous display/pan/zoom/rotate even with 10 million points.

Jean-Romain, 2017-09-30:
> For instance, being able to plot a las and then add a second las and maybe even **add the dtm** to the plot would be awesome for visulization purposes.

This is already possible. The current `plot` on `rgl` so it is possible. How? I'm not sure. Have you ever try `add = TRUE`?

bi0m3trics, 2017-09-30:
There's no (real) way to run cloud compare in R, except for using system to call the command line version... so it doesn't meet your needs but in a pinch it does allow one to automate some nice visuals. Plus it can't visualize a surface...

Sometimes I use the rgl's plot3d to plot points and grids together (and apply lighting)... I've just not done it enough to tweak your implementation but maybe it woukd be worth givibg it a go to add "add". As it stands, plot3d works ok, but as you said... doesn't work for large renderings. Hell, I've even used meshlab for some similar type applications....

I understand the purpose of your new package and think it's needed. It would just be nice to have a little bit of bells and whistles...

Jean-Romain, 2017-09-30:
I'm alway keen to make new features. But you must understand I just learned OpenGL this night to create the package this morning and make it reliable this afternoon. So for the moment a single feature is good enough (and I don't know how to make more :wink: ). 

A first release on github will be available tomorrow if I solve some boring problem of trigonometry with "pan".

bi0m3trics, 2017-09-30:
I completely understand!

Jean-Romain, 2017-10-01:
https://github.com/Jean-Romain/PointCloudViewer

closed 2017-10-30 by Jean-Romain