r-lidar/lidR #67: grid_metrics has weird center points

thatsciencegal, 2017-07-13:
I am using grid_metrics to calculate lidar metrics in a 1000x1000 m tile. I have the center at c(0,0). I have several different grid sizes, but some of the x,y coordinates are outside of my 1000x1000 m tile when I plot them into GIS (see image below for 100x100 m grid size).

Jean-Romain, 2017-07-13:
Hi,

I can't see any image in your post. So I guess your problem arise at the edge of your dataset.

`grid_metric` create a continuous set of pixel and looks in each pixel the lidar points which fall inside the pixel and computes a user defined function on those points no matter if the pixel has 1 or 1000 points. Pixels at the edge of your dataset may be partially populated by lidar points.

For example if the starting point of the grid is (0,0) and the grid size is 1000 x 1000 m then, the center of the first pixel center is (500 x 500), If your lidar dataset range from 800 on the X axis, for example, then the center of the first pixel is outside the lidar data and is only partially populated by points.

Whatever the methods to compute the pixel location you will always have ''incomplete'' pixels at the edges of the dataset.

If your problem does not arise at the edge of the dataset, then it is a real bug but I'm pretty confident your question is relative to the edge of the dataset.

thatsciencegal, 2017-07-13:
I'm not sure why the image isn't showing. The issue is arising at the edge, however, in a 1000x1000 m tile, subdivided into 100x100 m pixels, I shouldn't have extra rows (I don't think). My x- and y-axes are each 1000 m long, which should be easily divided by the 100 x 100 m pixels. However, I have an extra row of pixels on the y-axis, so my grid is 10 x 11 instead of 10 x 10. Here is a link to the image and to the resulting data set respectively

https://doi.org/10.6084/m9.figshare.5203942.v1
https://doi.org/10.6084/m9.figshare.5203984.v1

Jean-Romain, 2017-07-13:
I was right. Your problem arise at the edge of the dataset. It's not a bug. If you want to compute a continuous 1000 x 1000 raster over the entire lidar coverage you must rely on your own code to load the data in a way which enable to get lidar data that fall exactly in your pixedl. This feature is not yet implemented in the version 1.2.1 (on CRAN). However it is in development in version 1.3.0 available on github in the `devel` branch. With version 1.3.0 what you want to do can be written like that:

```r
myctlg = catalog("path to las files")
grid_metrics(myctlg, .stdmetric, 1000, filter = "drop_z_below 0")
```

This way, you will still have partial pixel at the egde of the entire dataset but not for each file. The output is continuous. If you want to try this new feature do not hesitate to communicate with me. It is still a feature in development.

In the previous example the normalization is not included. If you want to manipulate more complex script the function `catalog_apply` is also available but currently I cannot guarantee it will work without failing. Indeed the function is still in development.

------

Nothing related to your question but you should rewrite the following line:

```r
las_norm@data$Z[las_norm@data$Z <0] <- 0
```

like that:

```r
las_norm@data[Z  < 0, Z := 0]
```

And learn how to manipulate [data.table](https://github.com/Rdatatable/data.table/wiki). The reason I'm saying that is because lidar data are very large data and R is inefficient at manipulating large data. With the `data.table` you have efficient tools to manipulate large amount of data. This is why the `lidR` package relies on `data.table` and not on `data.frame`

Jean-Romain, 2017-07-13:
> The issue is arising at the edge, however, in a 1000x1000 m tile, subdivided into 100x100 m pixels, I shouldn't have extra rows (I don't think)

Maybe yes. If your dataset range on X axis from 800 to 1800 and the grid start at (0,0) i.e. a pixel center at (50, 50) for a resolution of 100 m, then, you will have 10 rows at  850,  950, 1050, 1150, 1250, 1350, 1450, 1550, 1650 and 1750. However if your dataset range from 825 to 1825 you will have 11 rows at 850, 950, 1050, 1150, 1250, 1350, 1450, 1550, 1650, 1750 and 1850. The last one being partially populated

thatsciencegal, 2017-07-13:
> However if your dataset range from 825 to 1825 you will have 11 rows at 850, 950, 1050, 1150, 1250, 1350, 1450, 1550, 1650, 1750 and 1850. The last one being partially populated

This would then be solved by figuring out where my x (or in this case, y) axis starts and changing my grid start from (0,0) to (0,25) for the above example?

Jean-Romain, 2017-07-13:
Exact. 

In your specific case, something like that should work fine

```r
minx = las_norm@header@data["Min X"]
miny = las_norm@header@data["Min Y"]

las_norm %>% grid_metrics(.stdmetrics, res = raster_sizes[j], start=c(minx, miny))
```

thatsciencegal, 2017-07-13:
Thank you so much for your help!

Jean-Romain, 2017-07-13:
My previous code is wrong. Since the parameter start is the left bottom corner of the grid not the center of the pixel. But you understood the principle :-)

thatsciencegal, 2017-07-13:
It worked with the exception that I cannot put `start = c(minx, miny)`. I'm getting an error that states I have to use the actual number value of minx and miny. I will need to figure out a work-around for that, as I'm currently looping through a large number of files and 12 different raster resolutions.

Jean-Romain, 2017-07-13:
What is the error ?

thatsciencegal, 2017-07-13:
`Error: Parameter 'start' should be numeric`

`minx` and `miny` were being stored as non-numeric values. I added

`minx <- as.numeric(minx)
  miny <- as.numeric(miny)`

which solved the issue.

Jean-Romain, 2017-07-13:
Ok thank you I will solve this problem internally as well.

closed 2017-07-13 by Jean-Romain