r-lidar/lidR #68: Missing Functions in rumple_index....?

mliskAL, 2017-08-01:
Greetings,

I first came across your work the other day while looking for the ability to calculate rumple index from a raster within R and the initial search lead me to your rumple_index function on GitHub Gist (https://gist.github.com/Jean-Romain/f5608862d1521a0e756134445c9e42ed). I have separate questions about that particular function, but was curious enough to begin looking at the other work you've posted to here. Eventually finding the rumple_index function as part of this package, I installed the package to give it a try. 

The first issue which came up was an error: "could not find function rumple_index_internal". After some searching of your R code on this site, the named function was found below the code for rumple_index. While strange that the function wasn't detected, I copied your code into a local script and continued. However, after trying this another error was generated: "could not find function tinfo", which as you'll recall is used to calculate 'N' which is use to subsequently calculated both 'area' and 'parea'. I was unable to find tinfo as part of this or your rlas package, and searching the Internet for a function by that name didn't produce any useful results. 

My most immediate question for you is: where does the tinfo function come from? However, I thought you'd like to hear the rest as the functions didn't work upon installation. Having built R packages myself, from looking at your code I can't explain why the rumple_index_internal function was not recognized by my system, but it is also clear we use different methods for organizing our respective packages. 

Should the information prove useful, I'm currently using R version 3.3.0, with my OS being Windows 7 Professional. 

Thank you for your time and any answers you might be able to provide,

Matthew D. Lisk

Jean-Romain, 2017-08-01:
Hi,

Thank you for your bug report.

> initial search lead me to your rumple_index function on GitHub Gist 

Forget the Gist version of the function. The Gist function is just a try. I like how the rumple_index is computed in this version but it does no correspond to any documented method.

> The first issue which came up was an error: "could not find function rumple_index_internal".

You installed the package yourself from Github on Windows. This is a known bug. Many people reported missing functions when they compiled the package on Windows themselves. This bug concern Windows only and self compiled code. This is non understandable to me and non reproducible even on a Windows machine. Install the version from the CRAN or install Linux (very good idea btw :smile: ).

The current version from CRAN is based on a Delaunay triangulation as explained in several peer-reviewed paper. However it is not a good method for many reasons when applied to a raster. The next release has another implementation. Since you can't install the package from the CRAN you can copy paste the source code (fom line 140): https://github.com/Jean-Romain/lidR/blob/devel/R/metrics_canopy_roughtness.r

Jean-Romain, 2017-08-02:
>  where does the tinfo function come from?

If the function is called without namespace it comes from R-base or it is an internal function of the package. `tinfo` is not an R-base function so it is an internal function of the package. Here it is a C++ function so it can be found in the `src` folder https://github.com/Jean-Romain/lidR/tree/master/src

closed 2017-08-04 by Jean-Romain

mliskAL, 2017-08-07:
Thank you for your quick reply. I had expected GitHub to send me a notification that you had replied, but it wasn't until deciding to check that I realized that that expectation was false. 

Anyway, yes, thank you for answering all of my questions. Initially the CRAN installation failed due to my running an older version of R. Having updated due to other reasons, I tried installing again and it did so without issue. The function also ran without issue. That being said, there didn't seem to be an easy way the adapt it to exactly what I was looking for. 

Having looked at your updates, I did go ahead and copy that code. Ultimately, what has been needed is a function that will calculate the rumple index _and_ be able to be used as the input function for the raster package focal function. Your updated code was easily adapted to that purpose. 

Thank you again for your time.

Jean-Romain, 2017-08-09:
Great. Could you please share your code. I'm interested to know how you solved your problem.

mliskAL, 2017-12-06:
Greetings,

Sorry for resurrecting this dead topic, but I hadn't seen your reply and request for code. Here is what I had come up with, should it still interest you:

treeHeight <- raster('height values)
rad <- 'some radius'

focalWin <- focalWeight(treeHeight, rad, type='circle')
focalWin[focalWin>0] <- 1

getRes <- res(treeHeight)
focalRumple <- raster::focal(treeHeight, focalWin, fun=function(x){rumple_index.matrix(as.matrix(x, byrow=T), getRes[1], getRes[2])}, pad=T)