r-lidar/lidR #69: Lastrees results null value

karnayogendra, 2017-08-07:
Hi Jean, 
I am trying to segment the individual tree crown using "li2012" algorithm of lastrees function but the results returned into the NULL, however, I am able to do the segmentation using "dalponte2016" and "watershed" algorithm. Here is the details code that I used.
`RLas <- readLAS("F:/RQ1_RQ2/processing/2007norm/07_ROI10_norm.las")`
`lidarfilter <- lasfilter(RLas, Z>=0 & Z <=50)`
`t2 <- lastrees(lidarfilter, "li2012")`
t2 returned into NULL.

Your help will be much appreciated.
Many thanks.
Regards,
Yogendra

Jean-Romain, 2017-08-07:
Hi, `lastrees` returns `NULL` because the original point cloud is updated by reference. The non NULL output are only *extra* output.

> **Value**
>
> Nothing, the point cloud is updated by reference. If extra = TRUE, "dalponte2012" returns two RasterLayers, "watershed" returns one RasterLayer and "li2012" does not support the extra parameter.

Jean-Romain, 2017-08-07:
Hi will change the example in the doc and in the wiki. You're not the first one to ask this question. I will remove the 'extra' data from all the examples.

closed 2017-08-10 by Jean-Romain