r-lidar/lidR #70: Review doc for v1.3.0

Jean-Romain, 2017-08-20:
* catalog_apply.r
* options.r
* catalog_query.r
* catalog_reshape.r
* grid_canopy.r
* grid_density.r
* grid_metrics.r
* grid_terrain.r
* grid_tincanopy.r
* plot.catalog.r 
* `rumple_index` in metrics_canopy_roughtness.r
* readLAS.r
* plot.lasmetrics.r

closed 2017-09-04 by Jean-Romain

spono, 2017-10-25:
Hi Jean-Romain,
please, may I ask you to provide even few lines of description for the `lasground`/`lasground_pmf` command?
I'm interested in testing it at different (submetric) window's sizes on TLS  data but, according to your implementation, it seems that the `cellsize` parameter is not used. Do `min_ws` & `max_ws` refer to meters or what?
Thank you in advance!

Jean-Romain, 2017-10-26:
Hi,

Are you talking about the current `devel` version? I re-designed some functions to prepare the package to integrate more algorithms. Your question is pertinent but to be honest some time I'm a little bit confused between what is currently in the released version and what is in the development version.

Initially the original PMF described by Zhang et al. works on a raster. Thus, you need a cell size parameter to tell the algorithm what is the resolution of your raster. Then the algorithm consists in applying morphological operations to the raster. In the `lidR` implementation the morphology is applied at the raw point cloud level. There is no raster an thus no cell size parameter.

I do agree that the algorithm is no sufficiently documented yet. But this will be improved in the next release.

If my explanation is not clear enough do not hesitate to ask me for more details.

spono, 2017-10-26:
Right now, I'm using version 1.3.2 .
What I don't understand is: if `cellsize` doesn't refer to a search "grid", what is the measurement unit of the parameters related to the incrementing window? (i.e. if I want to use it in millimeters, shall I simply set to 0.001?)
On ALS data I used almost the default values and it works well (thinking to the numbers as "in meters"), but reading better your description it seems that mine was a struck of luck.

Jean-Romain, 2017-10-26:
In version 1.3.2 the parameter `cellsize` still exists. It is used to compute the threshold  increment as in the original paper (equation 7).  It is not use for the rasterization but it is used in the formulas of the original paper. So its units is "meter" (or to be precise the units of the coordinates of your point cloud).

In the next release, this parameter will maybe disappear because it is not really useful. We can replace `s*c` (equation 7) by a single value.

Jean-Romain, 2017-10-26:
This is a subset of the source code. You can see that `CellSize is used both to compute the thresholds (equation7)  but also to compute the windows size. But this is not correct according to the original paper is corrected in the next release.

```r
while (window_size <= MaxWinSize)
{
    # Determine the initial window size.
    [...]
    window_size = CellSize * (2.0 * (iteration + 1) * base + 1)

    # Calculate the height threshold to be used in the next iteration.
    if (iteration == 0)
      height_threshold = InitDist
    else
      height_threshold = Slope * (window_size - window_sizes[iteration]) * CellSize + InitDist

   [...]
}
```