r-lidar/lidR #71: catalog_apply example and classifying points inside polygons

bi0m3trics, 2017-09-03:
JR - The new catalog functionality is awesome and I think may have found a bug with `catalog_apply `in devel. When I take your example, tweak the .shp file to have a second polygon inland (in the attached zip and read in by editing line 18 to read `lake_shp = rgdal::readOGR(folder, "lake_polygons_UTM17")[2,]`) and classify for inclusion as opposed to exclusion (change `== FALSE` to `==TRUE`; line 32), I get the following error:
```
No polygon found within the dataFALSE
 Show Traceback
 
 Rerun with Debug
 Error in is.data.frame(x) : 
  trying to get slot "data" from an object of a basic class ("NULL") with no slots In addition: Warning message:
No point found with the condition: ~lake != FALSE & ~Z > 4 
```

I consistently get this error when working with `catalog_apply `and `lasclassify` when trying to classify point within polygons but I'm not 100% sure it's not a bug in sp or rgdal with the SpatialPolygonsDataFrame. The attached .R is a simple version of the problem using your example code and data.

[extdata.zip](https://github.com/Jean-Romain/lidR/files/1273113/extdata.zip)

Jean-Romain, 2017-09-03:
Hi Andrew

`catalog_apply` still need extensive tests. It is not ready for production yet and contains several bugs for sure (but not easy to figure out where). It's nice you test it. However it does not seem to be a problem with this function.

Why did you write and what did you mean with:

```r
lake_shp = rgdal::readOGR(folder, "lake_polygons_UTM17")[2,]
```

This line failed on my computer. `lake_shp` is therefore not defined (NULL) further in the code and the functions failed to access to this variable. The proper code should be:

```r
lake_shp = rgdal::readOGR(folder, "lake_polygons_UTM17")
```

Without the `[2,]` the code run without error (but I think it exactly the example of the doc isn't it?)

bi0m3trics, 2017-09-04:
As I said, the functionality is awesome and I'm really impress with your work. Sorry if I got carried away using/testing it... once again, you've written code much better than the hacks I had been using to do something similar, so when I saw 1.3.0 I just dove in...

So in short, if you switch the bool on line 32 of your code to "===TRUE", say you want to classify and filter for points inside a polygon, you should get an error. This didn't make sense since the would using the lake point in lastrees so did the following

I took the shapefile from your example and added another polygon, not in the lake but on land (to minimize sources of error, I just edited your shapefile in QGIS and replaced the one in my lidR extdata folder so I could minimally change your example - it's in the zip I attached). You can just put it somewhere else rather than overwrite your version. This new shapefile now has two polygons (the lake one and the new one on land), so the [2,] is just subsetting the shapefile for only the land polygon (the second slot in the spatailpolygondataaframe).  I tried to make this example using all your example to make it easier... and maybe I juts made it more confusing.  In short, if you supply the catalog_apply a polygon you want to classify points within (as opposed to exclude like with the lake) it appears to throw errors. I can produce an completely different example if this still doesn't make sense.

Also (this is a joke) I asked Auty if I should comment on devel (as normally I would only comment on master) and he said yes so blame him!

Kind regards,
Andrew

Andrew J. Sánchez Meador, Ph.D.
Program Director of Biometrics and Forest Management, Ecological Restoration Institute
Assistant Professor of Forest Biostatistics and Quantitative Ecology, School of Forestry
Chair, Forest Science and Technology Board, Society of American Foresters
Northern Arizona University; Flagstaff, Arizona 86011-5018
Office: (928) 523-3448 * Cell: (928) 853-8650 * Fax: (928) 523-0296
E-mail: andrew.sanchezmeador@nau.edu
Web: www.for.nau.edu * www.eri.nau.edu * www.quantitativeecology.org

"You don't need anyone's permission to make something great."
                                                     - Massimo Banzi, TEDGlobal 2012
<http://www.eri.nau.edu/>

bi0m3trics, 2017-09-04:
The only other thing I've noticed thus far is that it looks like `lasnormalize `with `catalog_apply `should work with `knnidw `but the code for `catalog_apply `in the `devel `branch appears to exclude passing a value for a user supplied `res `argument (if you include, say `res=2`, you get the `unused argument (res = 2)` error message, and if you don't include it it works fine). If you want me to submit as a separate issue, just say the word... or if you want me to stop playing around with devel...   ;)

Jean-Romain, 2017-09-04:
Ok, you edited the shapefile by hand and replaced your own extdata folder. That is why you sent me the new shapefile and why your code still point to the extdata folder. I didn't catch up that. I'm gonna try again your example.

Yes, please open a separate issue with a reproducible example. It's impossible for me to deal with two potentially complex issues in a single topic.

And I'm not sure, we understood each other. It is nice you're trying the new functions. That help me a lot. When I said it is not ready for production yet, I meant *"do no be surprised if it fails"* but not *"do not use it"*

Jean-Romain, 2017-09-04:
Ok I reproduced it.

You created a very tiny rectangle and you tried to filter data inside this rectangle. However for 8 clusters over the 9 there is no-data at all inside the rectangle. Therefore `lasfilter(inland == TRUE)` returns NULL, your point cloud is empty. Then `lastrees` failed too.

It is both a bug and not a bug. I could design the functions to be more robust and enable them to fail with a more explicit message when fed with void data. However, as long as I allows to run user-defined function this kind of bug will exist forever.

bi0m3trics, 2017-09-04:
Ok and thanks. I figured it was both a bug and not a bug... I also think the responsibility is on the user to make sure the there are pointcloud data somewhere within their catalog+polygon combination, otherwise the analysis makes no sense. However, the scenario of a user supplying roi's that are within the catalog but have resulting tiles that don't overlap the provided polygon is highly likely. 

Your solution makes sense and I suggest just having a disclaimer in the documentation that there are no checks for validity, but at least this way it works for possible both scenarios and doesn't halt the process.

Thanks!

Jean-Romain, 2017-09-04:
> However, the scenario of a user supplying roi's that are within the catalog but have resulting tiles that don't overlap the provided polygon is highly likely.

The catalog and polygon overlap is the user problem. In `lasclassify` if you ask to classify your points based on polygons which are in another country,  the algo says the points are outside the polygons and the answer is correct. 

However the clusters (how the catalog is split in pieces) can overlap the catalog but being empty that is true. This case is managed internally and it should not create a bug.

> Your solution makes sense and I suggest just having a disclaimer in the documentation that there are no checks for validity

I agree. There are probably bugs in `catalog_apply` but I can't check bugs in the user-defined function. But I would like to provide better error message. This case was easy the object was `NULL`.

Cheers.

closed 2017-09-04 by Jean-Romain