r-lidar/lidR #73: lastrees: point cloud not updated

spono, 2017-09-05:
Hi Jean-Romain,
using the latest 'devel' version, Li2012 segmentation "works" (no warnings & progress bar runs till 100%) but the point cloud is not being updated with the field "treeID" .
Did something change in the process?

Jean-Romain, 2017-09-06:
Hi

Many things changed but not in this function. In addition there are a couple of unit tests in the package including tests for `lastrees`. The package currently pass all the test so it should work.

From my own experience with `data.table` sometime the `:=` operator fails if you made previous weird or wrong manipulations with your table. Re-run your code in a clean environment. If the problem persist, send me a reproducible example.

Cheers

spono, 2017-09-06:
No particular manipulations on <data.table>: I was using as input the LAS data extracted from <catalog_queries> and stored in a <list>. It seems working fine if I import directly from a <.las> file.
The code was working with the previous version. Possible sort of regression?

Jean-Romain, 2017-09-06:
I'm was necessarily talking about *your* wrong manipulations but maybe mine in internal code.

So the problem occurs with data extracted with `catalog_queries` but not when directly reading the las files. Ok this is a clue I'm investigating.

Jean-Romain, 2017-09-06:
Does this code work on your computer. It works on mine.

```r
folder <- system.file("extdata", "", package="lidR")
x <- c(684850, 684880)
y <- c(5017850, 5017880)
r <- 20
buffer <- 5
rnames <- c("plot1", "plot2")
 
ctg <- catalog(folder)
plots <- catalog_queries(ctg, x, y, r, roinames = rnames)

lastrees(plots$plot1, "li2012")
```

spono, 2017-09-06:
no, it doesn't work. Unfortunately I don't have the chance right now to test on another pc; I'll let you know as soon as I'm able to get more info.

Jean-Romain, 2017-09-06:
Well, actually it doesn't work on mine too. I think I caught up the issue. Try again with `catalog_options(multicore = 1)`.

```r
catalog_options(multicore = 1)

folder <- system.file("extdata", "", package="lidR")
x <- c(684850, 684880)
y <- c(5017850, 5017880)
r <- 20
rnames <- c("plot1", "plot2")
 
ctg <- catalog(folder)
plots <- catalog_queries(ctg, x, y, r, roinames = rnames)

lastrees(plots$plot1, "li2012")
```

I have the issue if the ROIs are extracted using several cores but not on single core. Please confirm the issue and I will try to figure out how this is possible.

spono, 2017-09-06:
I confirm: with single-core it works

Jean-Romain, 2017-09-06:
Did you say that the code works with the current CRAN version of the package? This bug is very weird but if it works with the previous version it is worst :worried: . I opened an issue on the `data.table` repo https://github.com/Rdatatable/data.table/issues/2333

spono, 2017-09-07:
I didn't use the [lidR] CRAN version but I always used the github one...and it was working till the last update I did few days ago.

Jean-Romain, 2017-09-07:
Sounds somebody understood the bug on data.table issues. There is no magic but the explanation under the hood implies some very deep stuff. I don't fully understand the problem but I roughly understand why it happens. However there is something I can't understand and which implies some kind of magic: why the hell it worked in previous release? It shouldn't.

Which one was your previous version? Do you remember the installation date? I need to install a previous version to make a try. Maybe the CRAN version can do the job.

Jean-Romain, 2017-09-07:
I pushed in `devel` some commits. One of them solve this issue in a dirty and memory inefficient way. However it is not a big problem since if you run `catalog_queries` using several cores you are likely to want to load several tiny ground inventories or something like and not several big tiles. Therefore it is not a big deal to get a memory inefficient line of code. Actually it is the normal R behavior to be memory inefficient.

But this is only a patch till the data.table's team find a real correction for this bug.

spono, 2017-09-07:
I've been using:
- 1.1.0 from CRAN
- the master version of 1.2.0
- then I moved to 1.2.1 `devel` when you solved the previous issue with `qhull` (#63) [reading from `ctg` was working].
- Finally, I updated it three days ago and things were not working anymore. Unfortunately I don't remember the previous version number.

So, I don't know if, in the meanwhile that you were publishing 1.2.1 on CRAN, something changed in it too...and unfortunately I really can't help on this.
let's hope to have some news from the `data.table` team

Jean-Romain, 2017-09-08:
Ok, I found a better way to solve the issue exploring the `data.table` documentation. I close the issue. You can install again from `devel`.

Thank you for reporting the bug. This was definitively tricky and difficult to find.

The question relative to why it worked in your previous version remains open. But actually... we don't really care :smile:

spono, 2017-09-08:
thank you for the great job!

commit b18b79cd (2017-09-08): alloc.col after parLapply (close #73)
closed 2017-09-08 by Jean-Romain