r-lidar/lidR #74: Time taken by lasground and grid_terrain commands?

ahmadshabbir2468, 2017-09-14:
I am beginner to pointcloud and R. I setup R in ubuntu and install LidR package. i want to classify Tree from las file. I am using steps given in "Tree-segmentation-from-A-to-Z". when i run lasground it takes long time and nothing happens same in the grid_terrain case. Anyone here tell me what is the reason behind it?

Jean-Romain, 2017-09-14:
Hi,

What do you mean my nothing happens ? The script crashed ? The computation is running over and over ? The computation ended but you don't know what to do with the result ?

ahmadshabbir2468, 2017-09-14:
`library(lidR)
las = readLAS("test.las")
plot(las)
lasground(las, MaxWinSize = 10, InitDist = 0.05, CellSize = 7)
dtm = grid_terrain(las, method = "kriging", k = 8)
plot3d(dtm)`

After readLAS, lasground commands runs and is not shifted to nextline of grid_terrain. I run lasground 2 hours ago and at this time they are at samepoint not shifted to nextline grid_terrain. :/

Jean-Romain, 2017-09-14:
I don't know what you mean by *"shifted to nextline"*. If you mean that the dataset is not normalized with the ground at 0 this is normal. `lasground` classifies the points as ground and not ground. `lasnormalize` enable to subtract the terrain. Two hours is long. I know that the function is not blazing fast but you may have a lot of points... Also the parametere are maybe not appropriated to your terrain. You're better to used the classification made by the data provider if available.

You should try your script with a small subset of your dataset before to apply it to big data. At least to test if it does what you expect it to do.

```r
library(lidR)

LASfile <- system.file("extdata", "Topography.laz", package="lidR")
las = readLAS(LASfile, Classification = FALSE)
plot(las)

lasground(las, MaxWinSize = 30, Slope = 2, InitDist = 0.01, CellSize = 7) = 7)
plot(las, color = "Classification")

dtm = grid_terrain(las, method = "kriging", k = 8)
plot3d(dtm)

las = lasnormalize(las, dtm)
plot(las)
```

Also notice that sometime the wiki pages are up-to-date with the development version but not with the released one.

closed 2017-09-14 by ahmadshabbir2468