r-lidar/lidR #75: Use for photogrammetry data?

Salman042, 2017-09-14:
Hi
I am working on photogrammetry data. Can i use this Package for pointcloud classification(classify buildings and tress). if there is any support of it for photogrammetry data please tell me. This package contains lot of good features and tools which help me to perform my operations on pointcloud data. 

Thanks.

Jean-Romain, 2017-09-14:
Hi,

`lidR` is designed to process point clouds. Photogrammetry is based on pictures so I think this package will no be very helpful for your purpose. But you have point clouds *and* pictures. So what kind of process would you like to do with your pictures and your point cloud? 

Regards

Salman042, 2017-09-14:
I have only example.las file which contains data as a lidar file contains but the classification column and return number in a file contain '0'. I just want to classify pointcloud data and separate out buildings and trees from it. I have not much knowledge about it how i can achieve this. So if you have any good guideline related to it please help me.

bi0m3trics, 2017-09-14:
I think he may be talking about Structure from Motion-based (some call it "phodar") point clouds, in which case you could process the resulting point cloud using lidR. You may run into some issues since you don't have information about returns (so no "ground" points) and the fact that sub-canopy penetration is difficult, the scan angle and gps time will be dummy variables (one single time for everything), however as long as the point cloud is structured using the asprs las/laz format you should be able to at least process/play woth it the data.

Also, some of the awesome functionality (say the Li et al algorithim for segmentation) may not make sense (versus a watershed segmentation), but as long as the data adhere to the las format you could load it and process it. Especially the RGB information that is usually already attached...

Andrew

Jean-Romain, 2017-09-14:
My mistake... photogrammetry! Photogrammetry generates point clouds

Well, I don't know. A point cloud is a point cloud. Building segmentation definitively not because there is no algorithm for that. Trees maybe. It depend on the orientation of the source. The algorithms implemented are made for point clouds acquired from sky. I can't swear they are suitable for other data but who knows...

bi0m3trics, 2017-09-14:
Agreed... I have a simple example I wrote up for a lab minimally processing a SfM point cloud derived from DJI Phantom 4 photos.... I'd be happy to share it and the data once I'm back in my office.

Andrew

Salman042, 2017-09-15:
Thank you so much for your informative reply. Its help me to understand the Lidar data more clearly. I am going to share the procedure how we generate Point cloud.
 we are using pix4d mapper software to create “.las” file which contains point cloud data. I am sharing “sample.las” file data screenshot so you can understand it more clearly that the las file we generate from pix4d mapper how actually looks like and data it contains:
[image]

This file only contains X ,Y , Z and R ,G ,B value.
@bi0m3trics  I am very thankful if you share something related to “Sfm point cloud” maybe its point me to correct direction.
Any Further suggestions related to this topic from your side help me a lot to further continue my work.
If you want more information related to this process you can ask and i discuss it more deeply.

Jean-Romain, 2017-09-15:
A las file is simply a convention for how to store point clouds. It contains X Y Z coordinates of each point as well as other scalar for each point independently of the source or the software. In your case all the ALS specific fields are empty since they do not makes sense for your source. 

You should display your point cloud instead of a table if you want to give us an idea of what it looks like. From a table it is not easy to guess anything.

```r
las = readLAS("file", Intensity = F, ReturnNumber = F, NumberOfReturn = F, ScanAngle = F, EdgeOfFlightline = F, ScanDirectionFlag = F, UserData = F, PointSourceId = F)
plot(las)
```

Or using the lastest develpement version from CRAN

```r
las = readLAS("file", select = "xyzRGB")
plot(las)
```

closed 2017-09-16 by Jean-Romain