r-lidar/lidR #76: catalog_index

floriandeboissieu, 2017-09-18:
Dear Jean-Romain,
what is the reason catalog_index was suppressed from the exported functions. it was very usefull to make intersection between plots coordinates and catalog. An lapply could be used, however as this function is still inside lidR code, I think it would be worth keeping it available from the outside.
Cheers
Florian

Jean-Romain, 2017-09-18:
Hi Florian,

`catalog_index` was designed to be used internally but was exported because *why not, it can be useful*. But actually it is a function which change relatively often as long as I improve the framework. It used to return a `data.table`, then a `list` and now on the `devel` branch it returns a list of S4 classes called `Cluster`. So far the new `Cluster` object is the most elegant and consistent way I found to create a consitent framework which is much more than *a buch of functions more or less consistant each other*.

So internally it changed a lot an often but externally it remains constant for the user. The framework is more and more robust and readable and usable for developers (me included*). This is why I can't export `catalog_index` for regular users. However if the framework become more stable in can be an option further.

Cheers

* If  you are lost in your own code it's time to write a better code

Jean-Romain, 2017-09-18:
To fulfill my previous answer, it took me 4 months to achieve the version 1.3.0 (and I didn't finished yet) but I was expecting to unexport the function sooner. 

Could you give me an example where this function is required? So far I think that `catalog_queries` is good enough for regular usages? And the new `clip*` functions will maybe replace `catalog_queries`.

floriandeboissieu, 2017-09-18:
An example would be:

1. I have a catalog of tiles

2. I have a data.frame of plots coordinates and field measurements on 
several sites

3. I would like to keep the plots only intersecting the tiles and 
eventually only the tile paths intersecting the plots to isolate them.

catalog_queries necessitate to extract the points. It will surely return 
NULL for plots out of catalog. So I could detect empty queries and 
remove them from plots list, however for just an intersection between 
tiles extents and SpatialPolygons/SpatialPoints the extraction is 
superflux. catalog_index was good for that, telling the intersection 
without loading any points.

But I understand that the evolution make it difficult to maintain for 
the moment, hope it comes back in further versions.

Cheers

Jean-Romain, 2017-09-19:
```
lidR:::catalog_index(...)
```

Returns a `data.table`in v <= 1.2.1, a `list` in v1.3.0 and a so far a `Cluster` in v > 1.3.0. In each case it return `NULL` for plot out of catalog but because the element `NULL` is nothing in a list the output list might be shorter that the input data. But the list is labeled so you always know in which plot you are.

The function is still documented and up-to-date `?lidR:::catalog_index`

closed 2017-09-21 by Jean-Romain