r-lidar/lidR #78: catalog_reshape: setting "human check" as selectable parameter

spono, 2017-09-26:
Hi Jean-Romain,
I'm using lidR to clip some point clouds out of a huge TLS scan.
I'm creating a catalog of the whole dataset (54 plots) and then, in a loop, I'm using reshape to cut them into more manageable files -> catalog again and then catalog_queries to extract the subplots.
Is there a way to automatically input the "1- yes" answer to the reshaping alg? or, would it make sense to set the "human check" as a boolean parameter of the alg itself? in this way, if included in a loop, it doesn't stop it just for checking.
all the best

Jean-Romain, 2017-09-26:
Hi,

It already exists but it is not documented:

```r
lidr_options(interactive = FALSE)
```
Why do you need to reshape in a loop? You can reshape an entire catalog at once. If your ultimate goal is to use `catalog_queries` you can also just use catalog queries on the big files. It is not a problem. If you want to speed up `catalog_queries` you can index your files with `.lax` files. see `rlas::writeLAX` or use `lasindex` from LAStools.

spono, 2017-09-26:
everything is meant to split the data in order to be managed always within the size limitations of R...but you're right: I can avoid the `reshape` to extract the ROIs.
How should be the `writeLAX` function be accessed: `rlas::writeLAX(myfile)`?
I'm also using the `filter` option directly in `catalog_queries` in order to ignore points above a certain threshold, but I noticed that expressions ( `filter="Z>min(Z)+4"` )are not allowed, right?

Jean-Romain, 2017-09-26:
> How should be the writeLAX function be accessed: rlas::writeLAX(myfile)

This is a function from the `rlas` package. The doc and the function can be accessed like any other function. You can call it via namespace (`rlas::`) or by loading the package (`library(rlas)`)

> I'm also using the filter option directly in catalog_queries in order to ignore points above a certain threshold, but I noticed that expressions ( filter="Z>min(Z)+4" )are not allowed, right?

Nope. This is an R expression. The filter is done at the C++ level. It cannot understand R stuff. The doc of `readLAS` states:

> The available filters are those from LASlib and can be found by running the following command: rlas:::lasfilterusage()

```r
readLAS(..., filter = "-drop_z_below 4")
```

Jean-Romain, 2017-09-26:
My mistake it is `writelax`.

spono, 2017-09-26:
amazing, thank you very much!

closed 2017-09-26 by Jean-Romain