r-lidar/lidR #79: One question

arecibo, 2017-09-28:
I guess that the entropy() function needs to be revised as follows:

entropy <- function(z, by = 1, zmax = NULL){
      if (is.null(zmax))
          zmax = max(z)
      if (zmax < 2 * by)
          return(NA_real_)
      if (min(z) < 0)
          stop("Entropy found negatives values. Returned NA.")
      bk = seq(0, ceiling(zmax), by)
      hist = hist(z, breaks = bk, plot = F)$count
      hist = hist/sum(hist)
      p = hist[hist > 0]
     ## pref = rep(1/length(hist), length(hist)) 
     pref = rep(1/length(p), length(p)) 
     S = -sum(p * log(p))/-sum(pref * log(pref))
     return(S)
 }

Jean-Romain, 2017-09-28:
Hi, thank you for your proposal. However I disagree with you. If you think you're right please give me some arguments. Here the explanation why it is computed like that.

Let suppose you have points at a single elevation slice. 17-18 m for example  (slices of 1 m) .

[image]

What is the entropy of this distribution? If you take a random points there are 18 possibilities where it could be (slice 0-1 m, slice 1-2 m, ...), but actually only one is actually realized. Whatever the random point we chose we know that the point belongs in slice 17-18 with a probability of 1. Thus the entropy is 0. You code computes `NaN` because you code take into account only the actual realizations not the overall possibilities.

arecibo, 2017-09-28:
Hi, 

Thank you for your quick reply.

Could you please take a look at the following code? 
The following code uses your entropy() function.

 R> x <- c(1, 1, 2, 2, 2)
 R> entropy(x)
 [1] 0.9709506
 R> y <- c(3, 3, 2, 2, 2)
 R> entropy(y)
 [1] 0.6126016

Since I did not read the book which is mentioned in the lidR package, I probably do not understand the concept of the normalized entropy measure. 
But I guess that the first entropy and the second entropy measures need to be the same. 
The main reason for the difference between the two is measuring ‘pref’ in the entropy function.

If I misunderstand the concept of the normalized entropy measure, please let me know.

Thanks.

Best regards,
Dong-hyun Oh

Jean-Romain, 2017-09-28:
According to a binning of 1m your first example has 2 bins possibilities (0-1 and 1-2) the second one has 3 possibilites (0-1, 1-2, 2-3).

In the first case the two possibilities are realized with a probability 2/5 and 3/5. The most random case (entropy = 1) is when the two  possibilities have the same probability 1/2 and 1/2. Here the information is a little bit less random so you have an entropy close to 1 but not one.

In the second case you have 3 possibilities but only 2 are realize. The most random case (entropy = 1) is when the 3 possibilities have the same probability 1/3, 1/3 and 1/3. But here it is not random at all. The probabilities are 0, 2/5 and 3/5. The information is less random, entropy is < 1.

In your interpretation you forget to take into account the location where we could have points but we don't. In your interpretation, That point distribution of point is almost perfectly random.

[image]

But it is not.

arecibo, 2017-09-28:
Let us assume that we gonna measure entropy of a character set.

x <- c(‘A’, ‘A’, ‘B’, ‘B’, ‘B’)
y <- c(‘C’, ‘C’, ‘B’, ‘B’, ‘B’)

Setting aside the fact that the entropy() function in the lidR does not measure an entropy of a character set, how do we map the vector x to a numeric vector in measuring entropy of x?
c(1, 1, 2, 2, 2) or c(3, 3, 2, 2, 2), or something else?

What about y?

Also we have an information that the number of bins of an alphabet string is 26. 
Then seq(1/26, 26) is pref?

My main objective of using the entropy() function of the lidR is to measure the entropy of a set of categorical variable. 
Hence, the mapping from a set of categorical variable into a numeric vector would be important.  

Best regards,

Jean-Romain, 2017-09-28:
Well, the entropy from lidR package is designed to compute the entropy of the point cloud distribution according to a given definition.

It is not necessarily suitable for every problem. In the example you wrote, `x`and `y` may have the same entropy. But if you consider the entire alphabet you're right `pref = seq(1/26,26)`. Does it have a meaning in your case? You alone know.

arecibo, 2017-09-28:
Many thanks.

closed 2017-09-28 by Jean-Romain