r-lidar/lidR #81: Grid_metrics: processing beyond the .las file extent

ovanlier, 2017-10-04:
Bonjour Jean-Romain,

I'm just getting my toes wet with your creation and am really enjoying it so far! So congrats on a job well done!

I have, however, run into a little issue with the output of grid_metrics that perhaps you could shed some light on... In brief, grid-metrics appears to produce outputs spatially outside of the .las file data extent. 

The picture below illustrates the issue. Red lines are the .las file extents and the green squares, the output from grid_metrics, and finally, the black lines represent a 20m fishnet. These outputs are generated based on small n (1-5).
[image]

Any insight would be most appreciated!
Thanks!
Olivier

Jean-Romain, 2017-10-04:
Hi olivander

This is normal and expected. The pixels outside the extend are expected to contains only few points. This is a rounding result. Let suppose that the las extent starts at x = 15 and your pixels are 20 m wide and the first pixel is at (0,0) (left bottom corner i.e a center at (10,10)). Thus the first pixels are those at x = 10, they contains point between x = 0 and x = 20. So actually points between 15 and 20.

So yes some dummy pixels can be outside of the extent. Remove them, they are just regular results from rounding using to a fixed starting point and a given length which is not a multiple of the extent size.

Jean-Romain, 2017-10-04:
Btw your display is nice. How did you do that.

ovanlier, 2017-10-04:
Thank you for your prompt response!

The display is in ArcMap. Export the grid_metrics output to .csv, add data in arcmap, right-click gridmetrics.csv --> display XY Data.. 

I have to disagree with you... unless I misunderstood the issue :D 
I think the pixels outside the extent should contain no points.

It has become common place to provide .las data for large extents in tiles. The origin of all tiled data must be placed on a whole meter coordinate value that will align with the zero origin of the UTM zone. The origin of this las file (bottom left) is 420000 5403000. All tiles are 1000m wide... hence there should be no rounding errors... Thoughts?

Jean-Romain, 2017-10-04:
I didn't know that. Do you have any kind of "official" documentation for that? From all the lidar data I encountered so far I can affirm this is rarely respected. But if you provide me some documentation about this practice I will change the starting point.

But you must understand it will not solve the problem. First there are files which are not UTM. There are subplot more or less randomy chosen. There are dataset not split as you described.  There are pixel sizes not multiple of 1 km. And on can find many case where what you say does not works.

Jean-Romain, 2017-10-04:
Well ok I misunderstood. It should work because the starting point is already (0,0) and your dataset is properly split in 1 km file as you described. Well ok so I will have a look one the code.

My first guess is that your pixels contain few points at the exact boundaries of the pixels.

Jean-Romain, 2017-10-04:
Below some results with my own dataset (properly split as you described). The extra pixels contain 1 or 2 points at the exact boundary. So they are expected to be on the next pixel. 

So the question is what are the official rule regarding boundaries inclusion. Does the left and bottom side are include or the right and top sides?

[image]

ovanlier, 2017-10-04:
First, thank you so much for taking the time to look into this this morning.

NRCan dirrective on ALS acquisition parameters was released recently http://ftp.maps.canada.ca/pub/nrcan_rncan/publications/ess_sst/304/304669/gip_117_en.pdf
but they do not mention (or at least from my brief overview) the las file starting on whole number 0 of UTM... But regardless, our data does :D

ovanlier, 2017-10-04:
Good point on the rules for inclusion.

Below, n=1 on left of red line (thus outside the extend of the LAS file), las xyz points with height 15.7m are the blueish triangles (15.7m was zmax, n=1)). It appears, that the one point near the boundary was used in this case, it's coordinates are 420000.040 5403176.330. It should therefore have not been included in the n=1 cell. 
[image]

Jean-Romain, 2017-10-04:
Well, even if you are right, your document start in the title page by "Natural Resources Canada". So it is specific to Canada. Maybe North America. I don't want to extrapolate beyond. So you problem is difficult. Here my answer:

* Your extra pixels contains few points at the exact boundaries. You understand that boundaries should be excluded on the left and bottom or right and top boundaries for continuity needs. I don't have a better explanation right now but I'm open to suggestions.
* So you necessarily have extra pixels on, at least, two sides if the file includes points on the edge
* You have extra pixels at the four egde. We must find why.
* You have an option to change the reference in the function

Jean-Romain, 2017-10-04:
> Below, n=1 on left of red line (thus outside the extend of the LAS file), las xyz points with height 15.7m are the blueish triangles (15.7m was zmax, n=1)). It appears, that the one point near the [...]

Well ok, it looks like compute precision error. But 0.040 it to big to be a computer precision error. I'm investigating.

ovanlier, 2017-10-04:
This is not always the case. Here, another example highlighting xyz points (triangle) with height matching the point used in this specific grid-metrics cell n=1. Note that no points are in proximity of the boundary. Why then are the gridmetrics computed for that empty cell...

I have to step out for a minute. When I return I will assess max/min XY of all xyz points in the las file to see if they go beyond the actual tile.

ovanlier, 2017-10-04:
[image]

Jean-Romain, 2017-10-04:
```r
lidR:::f_grid(c(420000.040, 420000.004, 420020.040), res = 20, star = 0)
> 420010 420010 420030
```

It is expect to be computed properly. Could you tell me again which point is wrong in you example. I'm a little bit lost in your super zoomed image.

ovanlier, 2017-10-04:
The later example, the gridmetrics returned an output with n=1 (green square) based on one xyz point with z=7.17. The little triangle is the las xyz data with z=7.17. It has coordinates of 420000.49 5403239.32. However I just looked and the scan angles do not match. It appears there are no point in close proximity to the cell for which we have gridmetrics with n=1. I'm not sure what data it used to compute the metrics in this case...

I can always call you if you like to discuss in more detail... just pm me your number..

Jean-Romain, 2017-10-04:
Show me your code

ovanlier, 2017-10-04:
#read in height/normalized LAS file
lidar = readLAS("F:/TEMP/CBSite2016-C3-ORT_4205403_21_0001_0001.las", all=TRUE)

#filter to only veg class above 2m and below 30m
lidar2m30 = lidar %>% lasfilter(Z >2 & Z<30, Classification == 3 | Classification == 4 | Classification == 5)

#Define function to calculate Mode of an attribute
Mode <- function(x) {
  ux <- unique(x)
  ux[which.max(tabulate(match(x, ux)))]
}    

myMetrics = function(X,Y,Z,Intensity, ScanAngle, ReturnNumber, Classification,dz = 1)
{
  metrics = list(
    SAMin = min(abs(ScanAngle)),
    SAMax = max(abs(ScanAngle)),
    SASd  = sd(abs(ScanAngle)),
    SAVar = var(abs(ScanAngle)),
    SAAvg = mean(abs(ScanAngle)),
    SAMode= Mode(abs(ScanAngle))
  )
  return( c(metrics, stdmetrics(X,Y,Z,Intensity, ScanAngle, ReturnNumber, Classification,dz = 1), entropy(Z, zmax=NULL, by=1),VCI(Z, zmax=20, by=1),LAD(Z, dz=1,k=0.5)))
}
metrics <- lidar2m30 %>% grid_metrics(myMetrics(X,Y,Z,Intensity, ScanAngle, ReturnNumber, Classification,dz = 1))

write.csv(metrics,file="F:/TEMP/Metrics.csv")

Jean-Romain, 2017-10-04:
Your code is good. And I like what you did this is exactly for this kind of usage I designed this function.

Well please share you file. We found the problem but there is still something weird we are not gonna find by speaking.

Jean-Romain, 2017-10-04:
speed up your code :wink: 

```r
myMetrics = function(X,Y,Z,Intensity, ScanAngle, ReturnNumber, Classification,dz = 1)
{
  sa = abs(ScanAngle)

  metrics = list(
    SAMin = min(sa),
    SAMax = max(sa),
    SASd  = sd(sa),
    SAVar = var(sa),
    SAAvg = mean(sa),
    SAMode= Mode(sa)
  )
  return( c(metrics, stdmetrics(X,Y,Z,Intensity, ScanAngle, ReturnNumber, Classification,dz = 1), entropy(Z, zmax=NULL, by=1),VCI(Z, zmax=20, by=1),LAD(Z, dz=1,k=0.5)))
}

ovanlier, 2017-10-04:
sent you email with ftp info for the las file

Jean-Romain, 2017-10-04:
Great thank you

Jean-Romain, 2017-10-04:
I did not receive anything yet. Is is normal?

ovanlier, 2017-10-04:
No, you should have received it through GitHub...

Jean-Romain, 2017-10-04:
No sorry

ovanlier, 2017-10-04:
I just forwarded it to your work email address

Jean-Romain, 2017-10-04:
The server is not responding

ovanlier, 2017-10-04:
Recently had someone else with same issue. Funny enough using FilleZilla (free download) worked...

Jean-Romain, 2017-10-04:
Good

Jean-Romain, 2017-10-04:
Here is your problem

```
min(lidar2m30@data$X)
[1] 419999.9799999999813735
```

Jean-Romain, 2017-10-04:
```
lasinfo CBSite2016-C3-ORT_4205403_21_0001_0001.las 
```

> Min X, Y, Z: 		419999.98, 5402999.98, -65.34
  Max X, Y, Z: 		421000.01, 5404000.01, 103.31
  Bounding Box:		419999.98, 5402999.98, 421000.01, 5404000.01

closed 2017-10-04 by Jean-Romain

ovanlier, 2017-10-04:
That's what I suspected- just wasn't familiar enough with the coding to figure that one out in R.
Thank you

Jean-Romain, 2017-10-04:
Btw your file is big do not load what you don't need into R. Use the streaming filter in readLAS.

ovanlier, 2017-10-04:
New to this, thanks for the hint.

ovanlier, 2017-10-05:
FYI, the source .las file from service provider has the exact extents (i.e. align with the zero origin of the UTM zone). After running the .las file through lasheight (LAStools), the extent grew by 2 cm on all edges (this is the file I had provided you). Now to streamline the process in R with lidR lasnormalize :)

Jean-Romain, 2017-10-05:
Ok the bug is from LAStools. Glad to know this is not a bug from by code :-)

> Now to streamline the process in R with lidR lasnormalize :)

Sorry, this is a question of english but I didn't understand this sentence. Do you want to replace LASheight by lasnormalize? You can but be careful. Your file is big. lidR is not as efficient as lastools.

ovanlier, 2017-10-05:
Yes, I was thinking of leaving out LASheight from the process and using lasnormalize from lidR, but you are making me second guess this...

Jean-Romain, 2017-10-05:
Try it n two step. create a dtm from ground point loading only ground points. Then load only point above 1.7m with classification = 3 as you previously did and subtract the dtm. This way you will never load the entire 1.5 GB.

ovanlier, 2017-10-05:
will do, thanks!

ovanlier, 2017-10-10:
FYI from lastools on lastools issue encountered:
Hello Olivier,

look in the control output in the console when you run this LAStools command. It will tell you that the software is "adding tiny xyz noise" to your output. That's what the license protection of the software is doing. Note that also your intensities, point source IDs, and GPS time stamps are zeroed. You can work around this by using much smaller tile sizes. This form of license protection is designed to allow folks to accomplish most of what they need for personal, educational, or non-profit research use while also giving incentive to those who use LAStools for production work to obtain a license. Sort of a balance act between allowing most everybody to use LAStools yet guarantee enough revenue for me to keep doing what I do ... (-:

Regards,

Martin @rapidlasso

Jean-Romain, 2017-10-10:
Now you said it, I knew it but I forgot it (actually I have never use lastools). Thank for the info.

rapidlasso, 2017-10-11:
Interesting discussion. Sorry it took so long for you, @ovanlier, to figure out but I thought that the control output in the console was unmistakably clear on the "tiny xyz noise". Nice package @Jean-Romain, I got a demo at IRD Pondicherry last week.

ovanlier, 2017-10-11:
Oversight on my end. Control output is clear on this: WARNING: unlicensed. over 1.5 million points. output slightly distorted. tiny xyz noise. points permuted. intensity, gps_time & point_source_ID zeroed.