r-lidar/lidR #83: Bug in catalog_makecluster

Jean-Romain, 2017-10-23:
This code creates a weird pattern of clusters. This is not a bug because internally the code handle properly empty cluster but it should be fixed.

```r
catalog_options(multicore = 1, tiling_size = 160, buffer = 0)

file <- system.file("extdata", "Megaplot.laz", package="lidR")
ctg = catalog(file)

m1 = grid_metrics(ctg, .stdmetrics, 20)
```

Jean-Romain, 2017-10-31:
The example in `catalog_apply` is bugged because of the error in the clustering

closed 2017-10-31 by Jean-Romain

spono, 2018-02-13:
hi JR,
using `catalog_apply` from `master` on my data, almost the same procedure (`multicore=3, tiling_size=100, buffer=0`) gives me `Progress: 100%` but in the task manager the 3 processes are still open, one of which running at 25-40% and the other two at 0%.
may it be the same bug?

Jean-Romain, 2018-02-15:
No it isn't ! If your problem persists is might be a problem with the `future` package imo. I can have a look but so far I didn't see any issue like that.

commit 5ef8886d (2017-10-31): Fix #83