r-lidar/lidR #84: lasscanline bug: .las$ScanDirectionFlag no supported

floriandeboissieu, 2017-10-24:
`$` access to LAS data is no supported anymore, 
`.las$ScanDirectionFlag` should be changed to `.las@data$ScanDirectionFlag`

Jean-Romain, 2017-10-24:
Yes because it was incredibly slow. And when I say incredibly I mean **incredibly**. Internally I changed some `las$X` to `las@data$X` and I speed up the algorithms by non negligible factors. So I removed this feature to prevent user writing slow and inefficient code.

But if you figure out a way to implement that in an efficient way to hesitate to tell me how.

Jean-Romain, 2017-10-24:
My mistake your are not questioning about code incompatibility you reported a bug... Right I missed one occurrence in `lasscanline`. I will fixed that.

Thanks

floriandeboissieu, 2017-10-25:
Hi Jean-Romain,

sorry for the misunderstanding, I was too late in lidR version to make a 
proper pull-request on devel branch.

Cheers

Florian

commit 73797bcd (2017-10-25): Fix #84
closed 2017-10-25 by Jean-Romain