r-lidar/lidR #85: [1.4.0.900] error in readLAS

spono, 2017-11-02:
hi JR,
I updated yesterday evening to `master`: loading a las I get this error:

`Error in .Call(`_rlas_lasdatareader`, ifiles, ofile, filter, i, r, n,  : 
  Incorrect number of arguments (14), expecting 12 for '_rlas_lasdatareader'`

Jean-Romain, 2017-11-02:
Hi,

Fun fact, I merged devel into master expecting you will make test ;-) (you or someone else but  actually you are the only one I know that use the non-released versions).

I think there is no bug. I think you must also update the `rlas` package. The question is: why R did not complain about the dependency versions?

Jean-Romain, 2017-11-02:
Well actually I don't understand how it could compile without the new version of `rlas`. I should not. It is impossible...

Please tell me what you did and give me a copy of the code used.

Thanks

spono, 2017-11-02:
ahahah "the lidR Guinea pig"

btw, I didn't use any strange code, just the same `readLAS` that I was using before.

The update, as well, happened using your code:
`devtools::install_github("Jean-Romain/rlas", dependencies=TRUE)`
`devtools::install_github("Jean-Romain/lidR", dependencies=TRUE)`

`rlas` is 1.1.6 and `lidR` is 1.4.0.900.

maybe a junk file persisted in the update process?

spono, 2017-11-03:
solved!
after reintstalling the packages, I removed them through `remove.package`. I cheked the library and `rlas` has been deleted while for `lidR` I found a `lidR.dll` that hasn't been removed.
i deleted it and used CCLeaner (just in case).
reinstalled everything and now it works.

Jean-Romain, 2017-11-03:
Great, it was probably a persisting junk file.

> ahahah "the lidR Guinea pig"

We usually say "beta tester". It is more "socially acceptable" :wink:

closed 2017-11-03 by Jean-Romain