r-lidar/lidR #86: safe version change

spono, 2017-11-10:
Hi JR,
is there a safe way to rollback to a previous version using `devtools::install_github`?
I uninstalled `master` from a pc to go back to back to 1.3.2 but on CRAN there's only 1.3.1: did it disappear?
Thank you!

Jean-Romain, 2017-11-10:
Everything is possible using git revert command and compiling yourself. Nothing disappear in git. But it is not easy. 1.3.2 just fixed a minor bug in the master branch after somebody reported the bug via email. I never tagged it as a version. There is no simple way to retrieve it. But 1.3.1 is very very similar.

What's wrong with 1.4.0?

spono, 2017-11-10:
the idea was to have my personal machine with 1.4.0 for testing and the workstation with a stable release.
Since I rolled back to 1.3.1, when I  covert the output from grid_canopy to raster using `as.raster` it results in the following error:
`Error in raster::rasterFromXYZ(x) : y cell sizes are not regular`.
used on the output from `grid_terrain` doesn't create any problem.

I'll have to evaluate if keep both machines with 1.4.0

closed 2017-11-12 by Jean-Romain