r-lidar/lidR #87: Weird cluster pattern from catalog_apply in 1.4.0.9

bi0m3trics, 2017-11-13:
This pertains to 1.4.0.9 (built using Github @20876a3 with rlas 1.1.6 and R 3.4.2) and I don't seem to encounter it with 1.3.1 from CRAN. Basically, I've been encountering some strange cluster patterns being built when using catalog_apply (I looked but I could not find this issue) and it seems to be an issue when the catalog is built form multiple tiles (thus the provided example uses catalog_reshape to split megaplot into multiple tiles and reads these in as a catalog) and doesn't happen with the catalog_apply example. 

Here's what the clusters look like:
[image]
...and then when processed it the output looks like this:
[image]

Here's a reproducible example:
[Example.zip](https://github.com/Jean-Romain/lidR/files/1465343/Example.zip)

I've encountered this issue with much larger data sets and it appears to always be the same. The lower left tile gets set up correctly but the remainder get the "striped" pattern.

Jean-Romain, 2017-11-13:
This is definitively very weird. However I don't have any trouble with your code on my machine. I solved a bug in the clustering function few days ago. I merged in master just now. Reinstall and retry. I think it should work.

bi0m3trics, 2017-11-13:
Hmmm... that's from a clean install from about three hours ago of everything (including R). But I'll redownload and reinstall lidR from github and see what happens, maybe I'll try it outside of Rstudio too, although I doubt that will change anything.

Jean-Romain, 2017-11-13:
I just push the fix an hour ago after you opened the issue. So yes it might solve the bug

bi0m3trics, 2017-11-13:
That solved it... thanks!

closed 2017-11-13 by Jean-Romain