r-lidar/lidR #88: [1.4.0.900] morpholodical opening error in lasground

spono, 2017-11-13:
Hi JR,
I still get the following error in `lasground`:
`Error in MorphologicalOpening(cloud$X, cloud$Y, cloud$Z, window_sizes[i],  : 
  Expecting a single value: [extent=0].`

Changing all the parameters, I understood that it does not to accept sub-metrical units. Indeed, it gives the error as long as I use a decimal number (e.g. `min_ws = 0.001`) but if I use whole numbers, it works.
Any idea? is it a matter of variable type?

Jean-Romain, 2017-11-14:
Please give me the code and parameters you used I will look at that.

spono, 2017-11-15:
` impLAS = readLAS(file, select="xyzi")
impLAS@crs = CRS(projectCRS)`
`lasground(impLAS, "pmf", min_ws=0.001, max_ws=0.5, min_th=0.01, exponential=TRUE)`

I first tried if it was related to the `exponential` equal T or F. No.
Then I tried changing the single parameters (mantaining decimals). No.
Finally I simply rounded them up and...it's working (maybe...no progress bar appered).

Jean-Romain, 2017-11-15:
Ok I reproduced the bug with the example data

```r
LASfile <- system.file("extdata", "Topography.laz", package="lidR")
las = readLAS(LASfile, select = "xyz")
lasground(las, "pmf", min_ws=0.001, max_ws=0.5, min_th=0.01, exponential=TRUE)
plot(las, color = "Classification")
```
I will investigate today and solve that before tomorrow. My first guess is that there is no real bug. But I should protect the code against bad inputs. You chose `min_ws=0.001`. So you mean 1millimeter. Less probable 1 millifeet but it does not change anything actually. How many points are contained within a 1 millimeter windows size? Probably only one, the central point. What does it means to apply a morphological operation with a single point?

I'm not sure this is the bug. But at least you should think about the question. I'm not sure your parameters are pertinents.

Jean-Romain, 2017-11-15:
You asked for a maximum windows size of 0.5. But equation (5) in original paper state the windows size are:

![](https://latex.codecogs.com/gif.latex?w_k%20%3D%202b%5Ek%20&plus;%201)

So for the first iteration k = 0 (exponential), b = 0.01 (mininum windows size) so w_k  = 3

The first windows size is 3 m and in the same time you stated the maximum windows size is 0.5. So nothing has been computed. The function entered in the C++ code with a NULL object and it crashed. I corrected that but it still does not work by design.

spono, 2017-11-16:
yes, sure, I definitely misundertood and I mixed up few things, like 'cells' and 'metres'.
I'll need then to understand how it was possible for the previous version to work...I'll try to check it out.

thanks as always for your time&availability!

Jean-Romain, 2017-11-16:
With you we always have the same problem. I should not work but it does...

Jean-Romain, 2017-11-17:
btw it is confusing. I will change that. I does not make sense to mix the original 'pixel windows size'. What I will do is enabling to provide any set of  windows sizes and any set of threshold and in addition I will provide supplementary function to compute windows sizes as in the original paper.

Jean-Romain, 2017-11-17:
PMF accepts now for user defined sequences. I thought about it and Zhang's parameter are really weak. There are many incoherences in the formulas. 

I added a function `util_makeZhangParam` to use the exact version of the original paper. This function raise errors and warnings for weird inputs that don't match or not match properly with the original equation.

This is much better, much robust and much flexible now.

Please tell me if the documentation look clear enough. I improved it as well.

spono, 2017-11-19:
I saw that it accepts whatever values, even fractions of the .las reference unit (metres/feet): great!
I'm still testing on the whole pointclouds but it seems to go fine.

The only thing that I may suggest, it would be to add (if any) the possible limitations to the `ws` and `th` parameters (e.g. "only >0", or "only odd numbers should be used").

but otherwise it is already great!

Jean-Romain, 2017-11-19:
This not restricted to odd number. Decimal number and even number are possible. But I will add a double check for negatives values.

closed 2017-11-19 by Jean-Romain