r-lidar/lidR #93: install.packages("lidR") -> udunits2 error

mkieser, 2017-11-22:
Do you have a solution to the following udunits2 error when installing the lidR package?

> install.packages("lidR")
Installing package into ‘/usr/local/lib/R/3.4/site-library’
(as ‘lib’ is unspecified)
also installing the dependencies ‘udunits2’, ‘units’, ‘sf’, ‘mapview’, ‘mapedit’

trying URL 'https://cloud.r-project.org/src/contrib/udunits2_0.13.tar.gz'
Content type 'application/x-gzip' length 67182 bytes (65 KB)
==================================================
downloaded 65 KB

trying URL 'https://cloud.r-project.org/src/contrib/units_0.4-6.tar.gz'
Content type 'application/x-gzip' length 783289 bytes (764 KB)
==================================================
downloaded 764 KB

trying URL 'https://cloud.r-project.org/src/contrib/sf_0.5-5.tar.gz'
Content type 'application/x-gzip' length 7734623 bytes (7.4 MB)
==================================================
downloaded 7.4 MB

trying URL 'https://cloud.r-project.org/src/contrib/mapview_2.2.0.tar.gz'
Content type 'application/x-gzip' length 1864996 bytes (1.8 MB)
==================================================
downloaded 1.8 MB

trying URL 'https://cloud.r-project.org/src/contrib/mapedit_0.3.2.tar.gz'
Content type 'application/x-gzip' length 20158 bytes (19 KB)
==================================================
downloaded 19 KB

trying URL 'https://cloud.r-project.org/src/contrib/lidR_1.3.1.tar.gz'
Content type 'application/x-gzip' length 897422 bytes (876 KB)
==================================================
downloaded 876 KB

Warning in strptime(xx, f <- "%Y-%m-%d %H:%M:%OS", tz = tz) :
  unknown timezone 'zone/tz/2017c.1.0/zoneinfo/America/Edmonton'
* installing *source* package ‘udunits2’ ...
** package ‘udunits2’ successfully unpacked and MD5 sums checked
checking for gcc... clang
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether clang accepts -g... yes
checking for clang option to accept ISO C89... none needed
checking for XML_ParserCreate in -lexpat... yes
checking how to run the C preprocessor... clang -E
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
**checking udunits2.h usability... no
checking udunits2.h presence... no
checking for udunits2.h... no
checking for ut_read_xml in -ludunits2... no
-----Error: libudunits2.a not found-----
     If the udunits2 library is installed in a non-standard location,
     use --configure-args='--with-udunits2-lib=/usr/local/lib' for example,
     or --configure-args='--with-udunits2-include=/usr/include/udunits2'
     replacing paths with appropriate values for your installation.
     You can alternatively use the UDUNITS2_INCLUDE and UDUNITS2_LIB
     environment variables.
     If udunits2 is not installed, please install it.
     It is required for this package.
ERROR: configuration failed for package ‘udunits2’**
* removing ‘/usr/local/lib/R/3.4/site-library/udunits2’
Warning in strptime(xx, f <- "%Y-%m-%d %H:%M:%OS", tz = tz) :
  unknown timezone 'zone/tz/2017c.1.0/zoneinfo/America/Edmonton'
ERROR: dependency ‘udunits2’ is not available for package ‘units’
* removing ‘/usr/local/lib/R/3.4/site-library/units’
Warning in strptime(xx, f <- "%Y-%m-%d %H:%M:%OS", tz = tz) :
  unknown timezone 'zone/tz/2017c.1.0/zoneinfo/America/Edmonton'
ERROR: dependency ‘units’ is not available for package ‘sf’
* removing ‘/usr/local/lib/R/3.4/site-library/sf’
Warning in strptime(xx, f <- "%Y-%m-%d %H:%M:%OS", tz = tz) :
  unknown timezone 'zone/tz/2017c.1.0/zoneinfo/America/Edmonton'
ERROR: dependency ‘sf’ is not available for package ‘mapview’
* removing ‘/usr/local/lib/R/3.4/site-library/mapview’
Warning in strptime(xx, f <- "%Y-%m-%d %H:%M:%OS", tz = tz) :
  unknown timezone 'zone/tz/2017c.1.0/zoneinfo/America/Edmonton'
ERROR: dependencies ‘mapview’, ‘sf’ are not available for package ‘mapedit’
* removing ‘/usr/local/lib/R/3.4/site-library/mapedit’
Warning in strptime(xx, f <- "%Y-%m-%d %H:%M:%OS", tz = tz) :
  unknown timezone 'zone/tz/2017c.1.0/zoneinfo/America/Edmonton'
ERROR: dependencies ‘mapview’, ‘mapedit’ are not available for package ‘lidR’
* removing ‘/usr/local/lib/R/3.4/site-library/lidR’

Jean-Romain, 2017-11-22:
That is a issue with your linux installation not with `lidR`

```
sudo apt install libudunits2-dev
```

closed 2017-11-22 by Jean-Romain

mkieser, 2017-11-22:
The installation issue was on a Mac.

resolved timezone with: https://stackoverflow.com/questions/4047188/unknown-timezone-name-in-r-strptime-as-posixct

then resolved gdal2 with: https://github.com/r-spatial/sf

then installed fine.

Jean-Romain, 2017-11-22:
Great ! Mac/Linux the same base. We call them unix-alike sytems