r-lidar/lidR #95: error with lasnormalize - output NULL

mkieser, 2017-11-22:
I've followed the doc to **load** a lidar dataset, build a **dtm**, and then **normalize** the lidar dataset based on the dtm but the result is "NULL".  [screen capture](https://prnt.sc/hdxpx9)

`lidar = readLAS("drawno.laz")`
`lidar`
class        : LAS (LASF v1.2)
memory       : 8.1 Mb
extent       : 278200 , 278300 , 602200 , 602300 (xmin, xmax, ymin, ymax)
area         : 9996.2 m² (convex hull)
points       : 150395 points,  107991 pulses
density      : 15.05 points/m²,  10.8 pulses/m²
field names  : X Y Z gpstime Intensity ReturnNumber NumberOfReturns Classification ScanAngle pulseID
coord. ref.  : NA

`dtm = grid_terrain(lidar, method = "kriging", k = 10L)`
[using universal kriging]
Warning message:
There were 3 duplicated ground points. Some X Y coordinates were repeated but with different Z coordinates. min Z were retained.
`dtm` 
  X        Y      Z
  1: 278200.5 602200.5 94.656
  2: 278201.5 602200.5 94.768
  3: 278202.5 602200.5 94.833
  4: 278203.5 602200.5 94.754
  5: 278204.5 602200.5 94.668
  9996: 278295.5 602299.5 93.513
  9997: 278296.5 602299.5 93.405
  9998: 278297.5 602299.5 93.389
  9999: 278298.5 602299.5 93.466
  10000: 278299.5 602299.5 93.467

`lnorm = lasnormalize(lidar, dtm)`
Warning message:
23155 points below 0 found.
`lnorm`
NULL

Jean-Romain, 2017-11-22:
I'm really sorry but I have to repeat myself.  The doc is very clear:

> The function returns NULL. 

So yes `lnorm` is NULL. It is not a bug it is a feature.

However `lidar` is normalized. Try `plot(lidar)`

mkieser, 2017-11-22:
> plot(lidar)
error: xp_attach_gl_context returned: 2
X Error of failed request:  0
  Major opcode of failed request:  149 (GLX)
  Minor opcode of failed request:  26 (X_GLXMakeContextCurrent)
  Serial number of failed request:  709
  Current serial number in output stream:  709

Jean-Romain, 2017-11-22:
That may be a real bug but it is really weird. I have never seen such bug in my whole R carrier. I need a reproducible example of your code as well as the original data (but if it is `drawno` I already have it).

What `Syst.setenv(TZ = 'MST')` means ?

Jean-Romain, 2017-11-22:
Ok I googled `error: xp_attach_gl_context returned: 2` and this sound to be OSX specific. 

But your screen shoot shows that `plot` works isn't it ?

mkieser, 2017-11-22:
TZ = 'MST' is just a reoccurring timezone issue with 'sf', trying to resolve separately

I think I got it:
`dtm = grid_terrain(lidar, method = "kriging", k = 10L)`
`lasnormalize(lidar, dtm)`

closed 2017-11-22 by mkieser

Jean-Romain, 2017-11-22:
Ok. 

Does `example(rgl)` works. If yes this is a bug in `lidR` if no this is a bug in `rgl` with OSX and `openGL`