r-lidar/lidR #97: Issue with V1.4.0.9 "Error: Incorrect argument 'filter'"

slpgnt, 2018-01-05:
Hi JR,

This may be a trivial issue that is unique to me. When running my code with v1.4.0.9 I received an error:
"Error: Incorrect argument 'filter'"
I narrowed it down to the readLAS function. For simplicity I tried loading in the MixedConifer.laz from the extra data attached to the package folder and received the same error. Below is the code used: 

> mixed <- readLAS("C:\\Users\\my_user\\Documents\\R\\win-library\\3.4\\lidR\\extdata\\MixedConifer.laz")
Error: Incorrect argument 'filter

Thank you

Jean-Romain, 2018-01-05:
Hi,

Are you Wim or Wim's colleague? Somebody wrote me by email about this issue yesterday and today.

This is a weird bug because `readLAS` do works perfectly (this is definitively the most reliable function tested several time per days...). It seems to be related to a bad installation of the package. They solved the issue by reinstalling the package.

Try and tell me.

slpgnt, 2018-01-05:
Hi,
I am not him nor his colleague, The function works when re-installing the old version of lidR. Installing the v1.4.0.9 package as below did not seem to work. The package appears to have installed correctly, but still gets the error.

library(devtools)
> remove.packages("lidR", lib="~/R/win-library/3.4")
> install_github("Jean-Romain/lidR")
> Reloading installed lidR
    lidR 1.4.0.9000. For help, for the most up-to-date version or to report a bug, visit https://github.com/Jean-Romain/lidR
library(lidR)
> mixed <- readLAS("C:\\Users\\cwear\\OneDrive - KBM Resources Group\\Documents\\R\\win-library\\3.4\\lidR\\extdata\\MixedConifer.laz")
Error: Incorrect argument 'filter'

Jean-Romain, 2018-01-05:
Which version of `rlas`are you using ? `utils::packageVersion("rlas")`

slpgnt, 2018-01-05:
Version 1.1.5

Jean-Romain, 2018-01-05:
Ok I reproduced the bug. I remember now. There is an incompatibility between `rlas 1.1.5` and `lidR 1.4.0`. This is why I must update `rlas` first before to release `lidR`. 

Install `rlas 1.1.6` from github.

slpgnt, 2018-01-05:
Great! Thank you for your help!

closed 2018-01-05 by Jean-Romain