r-lidar/lidR #98: "Problem readLAS"

santidilo2, 2018-01-05:
Error in .Call(`_rlas_lasdatareader`, ifiles, ofile, filter, i, r, n,  : 
  Número incorreto de argumentos (15), esperando 12 para '_rlas_lasdatareader'

Jean-Romain, 2018-01-05:
What was the context? Please tell me

- The code that generate this error
- The version of `lidR`
- The version of `rlas`
- Your OS

```
utils::packageVersion("rlas")
utils::packageVersion("lidR")
Sys.info()[['sysname']]
```

santidilo2, 2018-01-05:
Good morning, my problem is as follows.

I am developing a webapps using shiny Rstudio and some of the implemented tools are using lidR. Uploading our tools to shinyapps.io tells us that the lidR library is a "Bioconductor" package. The solution to the problem is to install the library directly from https://github.com/Jean-Romain/lidR, and it is installed perfectly. But when using, for example, the readLAS function I get the problem I left earlier.

This problem does not happen if the library is installed using install.packages ("lidR"), but when installed in this way we did not manage to upload our webapps to shinyapps.io. I would appreciate the solution ...

Jean-Romain, 2018-01-05:
I don't know shinyapps.io. I don't know how does it works and what are the rules on that platform. But `lidR` **is not** a Bioconductor package. It has one **suggested** dependency that is from Bioconductor. But you can install the package without this dependency.

Your problem is likely to be related to your version of  `rlas`. The current development version of the `lidR` package available on github requires the new version of `rlas`. But you didn't answered my question related to your versions of the packages so I can't swear this is the solution.

santidilo2, 2018-01-05:
The version packages:
rlas_1.1.5
lidR_1.3.1

Jean-Romain, 2018-01-05:
For lidR your version is 1.4.0 since you installed it from github. However there is an incompatibility between lidR 1.4.0 and rlas 1.1.5 but it is not expected to throw the error you showed to me. So this is weird.

Try to install rlas 1.1.6 from github on shynyapps.io.

santidilo2, 2018-01-05:
Thanks for the help and for the applications developed, they are great contributions. Installing rlas 1.1.6 plus lidR 1.4.0. directly from github was the solution for readLAS to work correctly.

Now for shinyapps.io,  the problem continues, it still says that it has problems using rlas and lidR. I hope to find the solution and if you find it I would appreciate it.

Jean-Romain, 2018-01-05:
What is the error reported by shinyapps.io?

santidilo2, 2018-01-06:
If I add to the code the lidR library when I upload the application to shinyapp.io, the following error appears:

An error has occurred. Check your logs or contact the app author for clarification.

If in the library code (lidR) is commented by #, then this error disappears and the graph is displayed.

The code is: 

library(shiny)
library(lidR)
library(rgl)
#devtools::install_github("Jean-Romain/lidR", dependencies=TRUE, force = TRUE)
library(rglwidget)

LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
lidar = readLAS(LASfile)
t = data.frame(x = lidar@data$X, y = lidar@data$Y, z = lidar@data$Z)
str(t)

ui <- fluidPage(
  rglwidgetOutput("plot")
)

server <- function(input, output, session) {
  output$plot = renderRglwidget({
    bg3d(col = "black")
    #str(data$d)
    #cols <- brewer.pal(11 , "Spectral" )[data$d$porosity]
    #cuts = cut(data$d$porosity, breaks = 35)
    plot3d(t$x,
           t$y,
           t$z, box = FALSE, axes = FALSE,
           aspect = "iso", xlab = "", ylab = "", zlab = "")
    rglwidget()
  })
}

shinyApp(ui, server)

Jean-Romain, 2018-01-06:
This does not display anything on my computer. But it's another problem.

closed 2018-01-06 by Jean-Romain