r-lidar/lidR #99: Node errors when catalog_options(multicore >= 2)

bi0m3trics, 2018-01-12:
Attached is a .zip file with four .laz files and code for a reproducible example of the issue (plus I thought you might like some more sw mixed-conifer data!). When I call `grid_terrain` or/and `catalog_apply` on a catalog (I got similar results when I built a catalog from Megaplot) the result intermittently returns a variety of `Error in checkForRemoteErrors(val) : X nodes produced errors; first error: ...` error messages for `catalog_apply `and never works for `grid_terrain`. I tested this and received the same outcome with `lidR 1.3.1` and `rlas 1.1.5` as well as with `1.4.0.9` and `1.1.6`, on both Windows 10 and on Linux Mint 18.3 with R 3.4.3 (and of course, with slightly altered code depending on the version I was testing).

However, when run in serial, everything works fine and produces perfectly viable results: e.g., http://quantitativeecology.org/files/maps/sfpeaks - this takes a few seconds to load...

[errorExample.zip](https://github.com/Jean-Romain/lidR/files/1626615/errorExample.zip)

Jean-Romain, 2018-01-12:
Well, I'm gonna suffer a bit with your examples :sob: .

In both cases  `grid_terrain` and `catalog_apply` sometimes worked and sometimes not on my computer with various `Error in checkForRemoteErrors(val)`. It will be extremely hard to debug a random error. 

In both cases I succeeded in running the code until the end. So it *does* work... But not always...

I guess the error comes from the communication between the processes. So it  might be a very deep issue far beyond my knowlegde and not related to `lidR`. But maybe it is my fault. It's difficult to say. But the fact that it works with a single core indeed indicates that the bug implies a lot of pain...

So I will study that but honestly I'm not super confident on my probability of success. But what you can do is to better configure the options. For example use `tiling_size = 400` instead of 100. Your sub-areas (clusters) will be 16 times bigger and thus you will have fewer clusters to process:

- You will process faster because the number of access to the files will be reduced
- You will process faster because the number of  'loop' will be reduced
- You will process faster because the number of overlap (computed twice) will be reduced 
- You will reduce the probability of bug by reducing the number of parallel loops.

Also I don't know how big is your entire dataset, but with the 4 files you provided you can load the entire dataset into the RAM. The size of the project once loaded into R is 140 MB. This is very small. The computation time will be better than with a multi-core process.

```r
project = catalog("~/Téléchargements/errorExample/")
project@crs<-sp::CRS("+proj=utm +zone=12 +datum=NAD83")
plot(project)
lidar = readLAS(project)
```

bi0m3trics, 2018-01-12:
Come on now - I have faith in your skillz!!! 

I do feel your pain, but I am glad it didn't always work for you in parallel either... so I'm not crazy. 😅  I'm also glad it's not just a Windows-related issue (e.g., good ole `clusterExport `) as I was scared my Linux-noob skills were off. I've started trying to figure it out on my end (obviously - more like playing really, but I'm genuinely searching and debugging) and will let you know if I come across anything useful... 

Regarding the example I sent - it was an example I created specifically for you for the bug submission. Thanks for the heads up on reading it directly into memory. I hadn't tried that (I have 96GB so I have room to play), but rest assured that I am using larger, almost optimized, tile_size values. Also, those data were only a 1.8% subset of the data I'm using for a project and decimated down from 16.8 pts/sqm! 

I've got an older script (pre-lidR) for a similar workflow that I used to use to get rLidar to work in parallel, so I'll go back and look at that to see if anything jogs my memory. Albeit, I'm such a lidR fanboy now I don't know if that code will make any sense to me anymore.

Jean-Romain, 2018-01-12:
> Also, those data were only a 1.8% subset of the data I'm using for a project and decimated down from 16.8 pts/sqm!

Ok, so your parameters seem more appropriated.

> I'm also glad it's not just a Windows-related issue

Actually to be cross platform and easy to maintain I use only MS Windows compatible multicore processes. I'm pretty sure we would not have such bug using forks instead of clusters. But this type of parallelization only works on real operating system :wink: 

> Come on now - I have faith in your skillz!!!

Could you explain me how to solve an issue which is random and not always reproducible? Seriously this is the main problem. Anyway, I'll investigate :smile:

Jean-Romain, 2018-01-13:
I think I solved the issue. At least it seems to work fine on my Linux machine. I use the package `future` to manage multicore processes. This is safer and all the complexity is manage with hight level functions.

- install the package `future`
- install `lidR` from the branch called `future`

Please tell me if it works fine on Windows too.

Jean-Romain, 2018-01-13:
So far I ran the code many times and I did not experienced any crash. In addition I added a visual progress information into the plot panel. Enjoy!

However now that stuff is managed by the abstraction of the package `future` the multicore mode is expected to change automatically as a function of the operating system. Thus on windows, what happens under the hood is expected to be very different than on Linux.

bi0m3trics, 2018-01-13:
Checking it now... I fell asleep way early last night. ;)

bi0m3trics, 2018-01-13:
YAY!!! Two Christmases in one month!??!??!?! 🎁 🎁 🎁  Seriously, that's an awesome solution and really elegant. 

I ran my posted example data and code multiple times in windows and it never crashed for me either. I then pointed it to 44 tiles (another subset ~2000ha) of the original acquisition (the higher density ones) and `grid_terrain` ran flawlessly multiple time. When I ran the new branch on `catalog_apply`, it threw errors which required me to tweak the default options for `future.globals.maxSize `(i.e., initially I received errors that the default settings in future were too small for the globals I was specifying so I did some "resources per core" math and reset the `options`) but fixing it was extremely easy once I read some of the documentation for `future`. I can easily see how, with caution, this facilitates the processing of landscapes of data. JR, it really is a nice addition as it allows the user to parameterize their runs even further (usually at their own risk!) and batch process entire acquisitions...

Great job man and thank you very much! Also, the addition of the "cluster progress" figure is a really nice touch/addition and that enhancement was not lost on me...it's hard not to watch the figure progress and not wonder what data/forest attributes are causing individual clusters to process slower or faster.

Jean-Romain, 2018-01-13:
Great so I'll merge the branch in devel and master. I'm fixing the last known bug and the v1.4.0 is almost ready (Dave is reading the doc).

Thank for reporting the issue and providing some runnable example.

closed 2018-01-13 by Jean-Romain