r-lidar/lidR #101: Running lasmetrics on multiple strata

slpgnt, 2018-01-15:
Hi JR,

This is a question.
Is there a way that one could run the Lastrees (li2012) on a catalog, then run a UD las_metrics function on multiple height strata (e.g. 2m or 4m ranges) on the individual tree? That is to run the segmentation once and recycle it through a Z filter to get the points within a strata.  I have tried various methods but the las_metrics function seems to require the lastrees be segmented within the same catalog function. You can image how time and memory consuming this task would be on a large catalog across 10-20 strata. I have racked my brain trying to solve this one but I'm sure it can be done.

Any help would be appreciated,
Thank you.

Jean-Romain, 2018-01-15:
Hi

Your question is something that is currently difficult to achieve. The segmentation result cannot be stored into a las file yet. Thus it is not possible to process an entire catalog which have been previously tree segmented. It's a pity. Once I will be able to write extra data into las files some new horizons might become available.

What is possible it to process your catalog sequentially piece by piece. For each piece (sub-areas) you segment the tree and you compute what you want to compute. You return your results one by one and you stack everything at the end.

This is what is described in the function `catalog_apply`. You can also find extra documentation [here](https://github.com/Jean-Romain/lidR/wiki/Apply-user-defined-analyse-over-an-entire-catalog).

Anyway this is computationally suboptimal for sure. But `lidR` is not designed to process large dataset. It is made to help you to try and explore some processes and algorithms on small to medium dataset. R is absolutely not the good language to process billions of points in a memory efficient way.

Jean-Romain, 2018-01-16:
FYI I fixed an important bug in `catalog_apply` just now. If you use it please update the package.

Jean-Romain, 2018-01-17:
`pulseID` is not a compliant LAS field. And `user data` is a good idea but the specs states that `user data` is an `unsigned char` (1 byte) so it cannot even store a simple integer (4 byte) . So no it doesn't work :disappointed: 

What is possible is to use extra bytes attribute. But so far it is not supported yet in write mode (in read mode yes but we still have some trouble with that).

slpgnt, 2018-01-17:
Ah, well thanks again for being on top of thing!

closed 2018-01-17 by Jean-Romain