r-lidar/lidR #103: Error in grid_metrics when processing catalog objects with more than one core

ptompalski, 2018-01-31:
When processing a catalog object with grid_metrics, the function fails if the multicore parameter in catalog_options() is set to >1. Example:

```
# create example data using megaplot.laz
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
LASfile2 <- LASfile

ctg <- catalog(c(LASfile, LASfile2)) #two objects inside the catalog

#this works:
catalog_options(multicore=2)
metrics <- grid_metrics(ctg, func = .stdmetrics_z)

# add custom metrics
myMetrics = function(z, i)
{
  metrics = list(
    zwimean = sum(z*i)/sum(i), # Mean elevation weighted by intensities
    zimean  = mean(z*i),       # Mean products of z by intensity
    zsqmean = sqrt(mean(z^2))  # Quadratic mean
  )
  
  return(metrics)
}

#this works:
catalog_options(multicore=1)
metrics <- grid_metrics(ctg, func = myMetrics(z = Z, i = Intensity))

#this does not work:
catalog_options(multicore=2)
metrics <- grid_metrics(ctg, func = myMetrics(z = Z, i = Intensity))
# Error: could not find function "myMetrics"
```

Jean-Romain, 2018-01-31:
Crap ! Another MS Windows feature... I know what is the problem but I can't reproduce the bug on a real operating system.

I changed the way multicore processes are managed. I was expected the new method to seamlessly manage this kind of thing providing a transparent interface working both on Windows, Mac and Linux. I was wrong.

The problem is that I cannot reproduce the bug and that I just released the new version of the package so I cannot fix the issue before a while :-(

bi0m3trics, 2018-01-31:
The provided script ran fine for me… 10 times in a row in windows on 3.4.3. What should I be seeing?

ptompalski, 2018-01-31:
Hi @bi0m3trics 
In my case I receive the error:
# Error: could not find function "myMetrics"
when the grid_metrics is used with more than one core. 

I am using 3.4.0 Patched, with lidR 1.5.0 (devel). I will update to 3.4.3 and check if that changes anything.

Jean-Romain, 2018-01-31:
@bi0m3trics answer disappeared. Maybe I did something wrong ??

I was starting to write to you Andrew to ask you for your experience. I released the package because I thought you tested it on windows. And I was right.

So the problem is not the one I'm thinking about. This is something else.

bi0m3trics, 2018-01-31:
I deleted the comment...  ;P I had to reduce the tiling_size (below ~200m to force clusters) before I actually got the error.

Jean-Romain, 2018-01-31:
Ok so it is what I was thinking. Bad new.

Jean-Romain, 2018-01-31:
Can you run that code

```r
library(future)

f = function(g)
{
  plan(multiprocess, workers = 2)
  o = list()
  
  for(i in 1:10)
  {
    o[[i]] <- future::future({g(i)})
  }
  
  return(o)
}

g = function(j)
{
  j + 1
} 

o = f(g)
o <- values(o)
```

ptompalski, 2018-01-31:
yes, this works fine. 
Retuned object `o` contains:

```
[[1]]
[1] 2

[[2]]
[1] 3

[[3]]
[1] 4

[[4]]
[1] 5

[[5]]
[1] 6

[[6]]
[1] 7

[[7]]
[1] 8

[[8]]
[1] 9

[[9]]
[1] 10

[[10]]
[1] 11
```

bi0m3trics, 2018-01-31:
same.
```
> str(o)
List of 10
 $ : num 2
 $ : num 3
 $ : num 4
 $ : num 5
 $ : num 6
 $ : num 7
 $ : num 8
 $ : num 9
 $ : num 10
 $ : num 11
```

Jean-Romain, 2018-01-31:
Ok I reproduced the bug. I'm trying to make a minimal reproducible example and I will move the discussion on the `future` repo.

I perfectly know the problem, and I clearly read in the documentation of  `future` that the question is managed by the package transparently. Because it also transparently use forks instead of clusters on Linux and because Andrew ran other codes properly I was sure that multicore was functional on Windows.

Andrew can you confirm that `catalog_apply` works fine ? Actually it should not because it relies on the same mechanism.

Jean-Romain, 2018-01-31:
I dug into the code and I understand the problem. I think I will have to rewritte some bunch of code. `future` do its job very well. This is how the call is made and how the expression is passed across the stack of functions that fails. Very very nasty bug 😞 Hopefully I have a flight tomorrow I will find time 😄

Jean-Romain, 2018-01-31:
Please try the `v1.4.2` in the `master` branch.

bi0m3trics, 2018-01-31:
I can confirm that catalog_apply works fine. I'm confident of that. However, I just ran the example code from the documentation on my home machine with 1, 2, and 3 cores and never saw an error; I also ran it with a catalog from single file (megalaz) and a catalog using two adjacent las tiles and all was fine. For what it's worth, I've run catalog_apply a ton since 1.4 and never had an issues or seen that mechanism fail.  

I then tried to run the grid_metrics example using myMetrics on a catalog using two tiles and I got the  error. 

Trying 1.4.2 in the master now...
[GridMetricsError.zip](https://github.com/Jean-Romain/lidR/files/1683719/GridMetricsError.zip)

bi0m3trics, 2018-02-01:
[image]
The bit of code in the above posted script works now... as well as the code that @ptompalski posted.

Jean-Romain, 2018-02-01:
Great. ✌️. I understand why `catalog_apply` works. The stack of calls is simpler. 

In `grid_*` functions used with a catalog, all the processes are driven by a single function `grid_catalog` which is extremely generic. This genericity is built upon a more complex stack of functions and it is not easy to propagate the user expression. Hopefully the `future` package is incredibly well designed.

closed 2018-02-01 by Jean-Romain